mikakaraila/node-red-contrib-opcua #680: node-red crash using ExtensionObjectArray

IW3IJQ, 2024-04-17:
Hello!
I'm using client node and I have to write a Variat array into an opc-ua server.
It's the same of case described [here](https://opcfoundation.org/forum/opc-ua-implementation-stacks-tools-and-samples/write-variant-array-badtypemismatch/).

The variable is /HostComputer/f078 (in previous there are the exact stucture) af a moulding machine.

Browsing this variable I get this:

```json
{
   "payload":[
      {
         "item":{
            "referenceTypeId":"ns=0;i=47",
            "isForward":true,
            "nodeId":"ns=2;i=115682",
            "browseName":{
               "namespaceIndex":2,
               "name":"Value"
            },
            "displayName":{
               "locale":"en",
               "text":"f078-Value"
            },
            "nodeClass":2,
            "typeDefinition":"ns=2;i=906090",
            "value":null,
            "dataType":"Variant"
         }
      },
      {
         "item":{
            "referenceTypeId":"ns=0;i=47",
            "isForward":true,
            "nodeId":"ns=2;i=115681",
            "browseName":{
               "namespaceIndex":2,
               "name":"State"
            },
            "displayName":{
               "locale":"en",
               "text":"f078-State"
            },
            "nodeClass":2,
            "typeDefinition":"ns=2;i=906032",
            "value":2,
            "dataType":"UInt16"
         }
      }
   ],
   "endpoint":"opc.tcp://xxx.xxx.xxx.xxx:4880/Arburg",
   "_msgid":"85269960dddcd3be"
}
```

my target is f078-Value ns=2;i=115682 and the type is ns=2;i=906090

Browsing custom types I get:

```json
{
   "payload":[
      {
         "item":{
            "referenceTypeId":"ns=0;i=45",
            "isForward":true,
            "nodeId":"ns=2;i=906034",
            "browseName":{
               "namespaceIndex":2,
               "name":"ArbVTypeNumber"
[... 489 lines cut ...]
    at /usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:214:26
    at Object.trigger (/usr/lib/node_modules/node-red/node_modules/@node-red/util/lib/hooks.js:166:13)
    at Node._emitInput (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:206:11)
    at Node.emit (/usr/lib/node_modules/node-red/node_modules/@node-red/runtime/lib/nodes/Node.js:190:25)
```

Checking the sources I seem to understand that typeId have to be filled in "Item" but there are no check about this.
If I add typeId

[image]

 I get a  crash with a different message:

```
12:51:22.892Z :populate_data_type_manager_104:41     Cannot find dataType Definition ! with nodeId =ns=2;i=906090
17 Apr 14:51:22 - [red] Uncaught Exception:
17 Apr 14:51:22 - [error] Error: Cannot find StructureType constructor for dataType ns=2;i=906090
    at DataTypeFactory.findStructureInfoForDataType (/home/nodetest/.node-red/node_modules/node-opcua-factory/dist/datatype_factory.js:91:15)
    at ExtraDataTypeManager.getExtensionObjectConstructorFromDataType (/home/nodetest/.node-red/node_modules/node-opcua-client-dynamic-extension-object/dist/extra_data_type_manager.js:46:45)
    at getExtensionObjectConstructor (/home/nodetest/.node-red/node_modules/node-opcua-client-dynamic-extension-object/dist/get_extension_object_constructor.js:24:39)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
    at async ClientSessionImpl.constructExtensionObject (/home/nodetest/.node-red/node_modules/node-opcua-client/dist/private/client_session_impl.js:1123:29)
    at async build_new_extensionObject_dataValue (/home/nodetest/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1751:36)
    at async write_action_input (/home/nodetest/.node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1741:11)
```

mikakaraila, 2024-04-23:
There seems to be null in some Datatype fields, I expect it will cause crash...

IW3IJQ, 2024-05-07:
Hello,
I was able to create a working dataset using raw node-opcua and write it to the server.
ExtensionObject is not involved.

This is the dataset read using node-opcua:

```
{ /* DataValue */
     value: Variant(
         Array<Variant>, 
         l= 2, 
         value=[
             Variant(
                 Array<Variant>, 
                 l= 9, 
                 value=[
                     Variant(Scalar<UInt16>, value: 1),
                     Variant(Scalar<UInt32>, value: 10),
                     Variant(Scalar<String>, value: Order Des),
                     Variant(Scalar<String>, value: Program Des),
                     Variant(Scalar<String>, value: Mould Name),
                     Variant(Scalar<String>, value: Material Des),
                     Variant(Scalar<String>, value: 07/05/2024),
                     Variant(Scalar<String>, value: Item),
                     Variant(Scalar<String>, value: Item Description)
                 ]
             ),
             Variant(
                 Array<Variant>, 
                 l= 9, 
                 value=[
                     Variant(Scalar<UInt16>, value: 2),
                     Variant(Scalar<UInt32>, value: 20),
                     Variant(Scalar<String>, value: Order Des 2),
                     Variant(Scalar<String>, value: Program Des 2),
                     Variant(Scalar<String>, value: Mould Name 2),
                     Variant(Scalar<String>, value: Material Des 2),
                     Variant(Scalar<String>, value: 07/05/2024),
                     Variant(Scalar<String>, value: Item2),
                     Variant(Scalar<String>, value: Item Description 2)
                 ]
             )
         ]
     )
   statusCode:      Good (0x00000000)
   serverTimestamp: 2024-05-07T16:20:18.000Z $ 000.000.000
   sourceTimestamp: 2024-05-07T16:20:18.000Z $ 000.000.000
}
```

And this is the object used in write action:

```javascript
    const order1 = new Variant({
            dataType: DataType.Variant,
            arrayType: VariantArrayType.Array,
            value: [
                    new Variant({
                            dataType: DataType.UInt16,
                            value: 1
                    }),
                    new Variant({
                            dataType: DataType.UInt32,
                            value: 10
                    }),
                    new Variant({
                            dataType: DataType.String,
                            value: "Order Des"
                    }),
                    new Variant({
[... 319 lines cut ...]
               "arrayType":"Scalar",
               "value":"Item2"
            },
            {
               "dataType":"String",
               "arrayType":"Scalar",
               "value":"Item Description 2"
            }
         ]
      }
   ],
   "topic":"ns=2;i=115682",
   "datatype":"ns=0;i=24",
   "browseName":"f078-Value"
}
```

I get the result "value written" but the client node wrote a null value: (Wireshark......)

[image]

using "raw" node-opcua:

[image]
[image]

PR #696 (2024-05-17): Ugly Variant Array support
Ugly code to fix #680 

Injecting this object the upc-ua node write values into opc-ua server as expected:

```json
{
   "_msgid":"64b2b4972106c468",
   "payload":{
      "value":[
         {
            "dataType":"Variant",
            "arrayType":"Array",
            "value":[
               {
                  "dataType":"UInt16",
                  "arrayType":"Scalar",
                  "value":1
               },
               {
                  "dataType":"UInt32",
                  "arrayType":"Scalar",
                  "value":123
               },
       [... cut]
closed 2024-05-17 by mikakaraila