mikakaraila/node-red-contrib-opcua #682: Support for displayName

Mradvanyi123, 2024-04-22:
Hello

Can the OpcUa-Server node be extended to allow the setting of display names?
Similarly, as the description is set. I tested the following for adding variables. 

`let displayName = "";`
...
```
 const dn = msg.topic.indexOf("displayName=");
  if (dn > 0) {
    displayName = msg.topic.substring(dn + 12);
    console.log(displayName);
    if (displayName.indexOf(";") >= 0) {
      displayName = displayName.substring(0, displayName.indexOf(";"));
    }
  }
```
...
`displayName: displayName || browseName,`

mikakaraila, 2024-04-26:
I added needed code, will come out in next version...

mikakaraila, 2024-04-29:
PLease test, new version available.

closed 2024-10-14 by mikakaraila