mikakaraila/node-red-contrib-opcua #683: OPC UA Node identifier

johnbarbosas, 2024-04-24:
Hello there, idk if it's an issue, but how can I access the OPC UA nodes using a string identifier, instead of using the number ID? Cause when edit my AAS on AAS Package Explorer and export this as a nodeset, the ID number changes every time, so even though I make a table including names and IDs, I need do remake this table every time I make a change on AAS.

mikakaraila, 2024-04-26:
I added this feature to read, which action needs this?

[image]

[image]

Code will detect that msg is not containing nodeId, it will translate browsePath back to nodeId and use it then... should be compatible with existing flows.

johnbarbosas, 2024-05-08:
I need to write in this value in the print, but I can't use "ns=6;i=128", because it changes.

[image]

DSousa21, 2024-05-08:
I'm having an error when applying this solution, "TypeError: Cannot read properties of undefined (reading 'length')" what does this mean and what am I doing wrong?

johnbarbosas, 2024-05-15:
I've already updated the pallete, and tried a lot of things, but always get the same error:

"warning: properties like messageType, namespace, variableName or VariableValue is missing."

I use this command: br=/0:Objects/6:AASROOT/6:AssetAdministrationShell:myStation1/6:Submodel:Handling2/6:Feedback/6:Q_94_Claw_2_Cart/6:Value

Idk, but I think numbers refers to namespace and names refers to BrowserName/DisplayName (tried both, didn't worked).

[image]

mikakaraila, 2024-05-17:
Should work as I have tested with Prosys UA Simulation server this:
[image]

mikakaraila, 2024-05-17:
Sorry there was typo in verbose logging... fixed to next version

mikakaraila, 2024-05-17:
Please test, I just published new version

johnbarbosas, 2024-06-05:
It worked, thxx!! Soon I'm gonna post here my flow.

closed 2024-06-06 by mikakaraila