mikakaraila/node-red-contrib-opcua #684: OPC UA subscribe "E_CONNSTATE_LOST"

SamReub, 2024-04-26:
Hi,

I've build a flow using the subscribe "multiple" functionality of the OpcUA client node. The node works fine, data in coming in at the right interval but I every 5s I get error "E_CONNSTATE_LOST" in the debug window. I can't seem to find this node in the current flow either... (debug panel is set to current flow and no other blocks besides this OPCUA flow). This looks like some kind of timeout in the connection, is there a way to solve this?

[image]
[image]

Thanks a lot!

mikakaraila, 2024-04-26:
Do you have delay that client node will have time to first connect to server before you try to subscribe item(s)?

SamReub, 2024-04-26:
No I did not add a delay (my flow is exactly like shown in the screenshot). I only have the OPCUA client node (Subscribe mode) so I don't see where I can add a delay?

mikakaraila, 2024-04-29:
Inject node, some seconds after start.

SamReub, 2024-04-29:
Issue remains, 10second delay implemented

mikakaraila, 2024-05-07:
Can you attach minimal flow to repeat & investigate problem?