mikakaraila/node-red-contrib-opcua #685: Support for BrowseName

daviddhauwe, 2024-04-28:
I want to create the following folder/variable structure

MainFolder - Folder1 - Variable1
MainFolder - Folder2 - Variable1

When adding folders/variables it is not possible to add a variable ns=1;s=Variable1 because the naming must be unique
So I add using ns=1;s=Mainfolder.Folder1.Variable1 which gives unique names.
However the browsename then becomes Mainfolder.Folder1.Variable1 which does not look like it should be with an OPC UA client. Moreover, my scada system has problems handling it.
The browsename should be Variable1 and the nodeid ns=1;s=Mainfolder.Folder1.Variable1
So similar to #682  I would like to make te browsename changeable.
I did already a test and also noticed that internally the browsename is used to keep track of the variables. Which then points to the same variable, so this should be changed to the identifier.
I will do some more tests and make a pull request when done.

mikakaraila, 2024-04-29:
Please test, new version available.

daviddhauwe, 2024-04-29:
Thanks for the quick response.
With the parallel introduction of the displayName, there is a small issue.
When the displayName is not set in the topic it defaults to the old browseName and not to the browseName set in the topic.

mikakaraila, 2024-04-30:
Latest PR merged, should be OK now

closed 2024-04-30 by mikakaraila