mikakaraila/node-red-contrib-opcua #690: Timeout issue

archerixx, 2024-05-03:
After latest update, I am getting following error message:
"Client node error on:  error: [" error at active reading: Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)"]"

This is example of my flow:
[opc_ua.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/15200117/opc_ua.json)

Worked just fine before updates 2 weeks ago.

mikakaraila, 2024-05-07:
Hmm, I fixed partly your flow so it works now:
[image]

Separated adding nodeIds and repeat readmultiple:
[image]

archerixx, 2024-05-07:
OPC server was bugged, seems that timeout happens now only from time to time. It is probably server's fault.

Is there limit to how many requests at a time can be made towards one server?

mikakaraila, 2024-05-07:
If your server is made with all features that specification contains then it should have diagnostics (UaExpert can show).
And the latest specification contains also server limits.

archerixx, 2024-07-01:
Hi,

Could you make this error/warning as one that can be caught with "Catch" node ?

Currently it keeps popping in debug and console and I can't do anything with it, so it would be great to catch it with Node-red's Catch node (so I can report in my system as device state)

Thanks

mikakaraila, 2024-07-02:
Sorry, but the error comes from the node-opcua... I expect that I cannot catch it.