mikakaraila/node-red-contrib-opcua #693: how to add multiple items in function at one time and read the values at one time

mingxu2008, 2024-05-05:
In the example, I found the function of adding multiple items in function at one time and subscribing, and whether it is possible to add multiple items in function at one time and read the values at one time

mikakaraila, 2024-05-07:
Check OPCUA-TEST-NODES.json from examples and this TAB:
[image]

archerixx, 2024-05-07:
Heya @mingxu2008. If you have names of all registers, you can use this flow. 
[OPCUA-ISSUE-690.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/15233649/OPCUA-ISSUE-690.json)

I made an array of registers in Inject node, they go once trough node.send() loop in function, then you just read them with "readmultiple" topic. This way worked great for me and seems to take least resources.