mikakaraila/node-red-contrib-opcua #694: OPC UA Client Configuration from a custom webpage. {It is not a Issue/Bug}

animeshkala9, 2024-05-06:
I want to collect Endpoints, Username, and Password from a custom local webpage. From the webpage, the user can add the Endpoint URL and other connectivity details without opening Node-Red dashboard.  I want to keep Node-Red under the hood.
Is it possible to do it on Node-Red?

mikakaraila, 2024-05-07:
Should be... I will be on vacation, but I will take a look afterwards.

animeshkala9, 2024-08-23:
@mikakaraila Hi, Have you had time to look into this? 
I want to make a Web Dashboard, where I would like to add some input field where user can add connectivity details ( Endpoint URL, Username, Password etc. 
I mean I want to send the Endpoint URL, username, password via an Inject node Payload. 

Please let me know if somehow we can do this using Node-red .. 

Thank You..