mikakaraila/node-red-contrib-opcua #697: How to renew the sourceTimestamp of an item when its value remains unchanged.

wyplzx, 2024-05-15:
hi, dear sir,
   I encountered a problem while using OpcUa-Server. The value and state of a item remain unchanged for a certain period of time, but its sourceTimestamp needs to be updated. The following message was sent to OPCUA Server, and through observation by UA-Expert, the "Source Timestamp" in the UA Expert window was not updated.
   May I ask for your advice for what message needs to be constructed to update the value of the "Source Timestamp" of item when the value unchanged.
   Thank you very much for taking the time to reply amidst your busy schedule. Thank you very much.
------------------------
Environment version: 
node.js:v16.20.2
node-red:v3.1.9
node-red-contrib-opcua:v0.2.328
msg send to OpcUa-Server：
    var newMsg ={
        payload: 10,
        topic: "ns=1;s=TestDevice",
        sourceTimestamp:new Date(1715335200000),
        datatype: "Float"
    };
return newMsg;
------------------------
2024/5/15 15:04:29[node: debug 181]
ns=1;s=TestDevice : msg : Object
object
payload: 10
topic: "ns=1;s=TestDevice"
sourceTimestamp: "2024-05-10T10:00:00.000Z"
datatype: "Float"
_msgid: "c13c4646c33aa2a3"

mikakaraila, 2024-05-17:
Use new Date() not fixed integer value

wyplzx, 2024-05-18:
Dear sir, 
    According to your guidance, the message construction method has been modified as following code (newMsg). 
    But after the modification, it seems that there were no new changes.
     My testing scenario is shown in the picture. 
     When sending messages with the same value to a node at different times, the sourceTimes and serverTimestamp seen on the client remain unchanged. 
    The attachment is a document for the process. 
    Thank you very much.

the message construction method:
var newMsg = {
    payload: 1,
    topic: "ns=1;s=MyDevice=1.MyEquipment=1.TestDevice=1",
    sourceTimestamp: new Date(),
    serverTimestamp: new Date(),
    datatype: "Float"
};
return newMsg;

[Test_SVR_flows.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/15359871/Test_SVR_flows.json)
[image]

IW3IJQ, 2024-05-21:
Try to put the timestamp in msg.timestamp;

the following code should be the one that handles the request:

https://github.com/mikakaraila/node-red-contrib-opcua/blob/67f36b48fe1d2b413f84018a5ab69acfed4baec3/opcua/102-opcuaclient.js#L1943-L1945

wyplzx, 2024-05-22:
Hello,  IW3IJQ,
      Thank you very much.
      And I attempt to add parameter: timestamp in the message. It was found that it did not work, and the effect was similar to before.
Meanwhile, I have another question: how can I update the serverTimestamp time of Node.

newMsg = {
    payload: 1,
    topic: "ns=1;s=MyDevice=1.MyEquipment=1.TestDevice=1",
    timestamp: new Date(),
    datatype: "Float"
};
return newMsg;

[image]

wyplzx, 2024-05-23:
Regarding the definition of the values of sourceTimestamp and serverTimestamp, based on protocols (7.7.3 SourceTimestamp) and (7.7.4 ServerTimestamp).
Does it mean that sourceTimestamp is determined by the source device, while serverTimestamp is mainly updated by the OPC UA server. 
When OPC UA server receives a new message, OPC UA server need to update servertimestamp to the current timestamp?

7.7.3 SourceTimestamp
https://reference.opcfoundation.org/Core/Part4/v104/docs/7.7.3

7.7.4 ServerTimestamp
https://reference.opcfoundation.org/Core/Part4/v104/docs/7.7.4

7.7.3 SourceTimestamp
The sourceTimestamp is used to reflect the timestamp that was applied to a Variable value by the data source. Once a value has been assigned a source timestamp, the source timestamp for that value instance never changes. In this context, “value instance” refers to the value received, independent of its actual value.
The sourceTimestamp shall be UTC time and should indicate the time of the last change of the value or statusCode.
The sourceTimestamp should be generated as close as possible to the source of the value but the timestamp needs to be set always by the same physical clock. In the case of redundant sources, the clocks of the sources should be synchronised.
If the OPC UA Server receives the Variable value from another OPC UA Server, then the OPC UA Server shall always pass the source timestamp without changes. If the source that applies the timestamp is not available, the source timestamp is set to null. For example, if a value could not be read because of some error during processing like invalid arguments passed in the request then the sourceTimestamp shall be null.
In the case of a bad or uncertain status sourceTimestamp is used to reflect the time that the source recognized the non-good status or the time the Server last tried to recover from the bad or uncertain status.
The sourceTimestamp is only returned with a Value Attribute. For all other Attributes the returned sourceTimestamp is set to null.

7.7.4 ServerTimestamp
The serverTimestamp is used to reflect the time that the Server received a Variable value or knew it to be accurate.
In the case of a bad or uncertain status, serverTimestamp is used to reflect the time that the Server received the status or that the Server last tried to recover from the bad or uncertain status.
In the case where the OPC UA Server subscribes to a value from another OPC UA Server, each Server applies its own serverTimestamp. This is in contrast to the sourceTimestamp in which only the originator of the data is allowed to apply the sourceTimestamp.
If the Server subscribes to the value from another Server every ten seconds and the value changes, then the serverTimestamp is updated each time a new value is received. If the value does not change, then new values will not be received on the Subscription. However, in the absence of errors, the receiving Server applies a new serverTimestamp every ten seconds because not receiving a value means that the value has not changed. Thus, the serverTimestamp reflects the time at which the Server knew the value to be accurate.
This concept also applies to OPC UA Servers that receive values from exception-based data sources. For example, suppose that a Server is receiving values from an exception-based device, and that
the device is checking values every 0,5 seconds,
the connection to the device is good,
the device sent an update 3 minutes ago with a value of 1,234.
In this case, the Server value would be 1,234 and the serverTimestamp would be updated every 0,5 seconds after the receipt of the value.