mikakaraila/node-red-contrib-opcua #699: OPC UA via VPN ReadMultiple Error

TobiasKorn, 2024-05-22:
Hallo Mika,

I am using your nodes (actions READMULTIPLE and WRITE ) to connect to an OPC UA server (Siemens Open Controller), which is in a different country via VPN. When the Open Controller was still in our factory we had no problems with the connection. Since it arrived the customer the READMULTIPLE is not working properly (Ping of the connection is about 90ms).

After NR restart the nodes show almost every time the error "invalid endpoint", later they both connect. But while the WRITE node is working, the READMULTIPLE node shows the error

`error: ["SessionX error at active reading: Transaction has timed out ( timeout = 15000 ms , request = ReadRequest)"]`

At first data request READMULTIPLE node returns normally a msg with an empty payload property, and shows after the second data request the error above. Transport settings are also used (MaxChunk = 8, MaxMsgSize = 8192, Send/ReceiveBufferSize = 1024). The amount of data read is between 30 to 240 BYTE.

Do you have an idea what wents wrong, or what else I can test?

Attached see also the NodeRed Server Logs.

Greetings
Tobias
[Serverlogs.log](https://github.com/mikakaraila/node-red-contrib-opcua/files/15404257/Serverlogs.log)

mikakaraila, 2024-05-23:
What is your read interval? If it is more than timeout 15 sec it will cause this...

TobiasKorn, 2024-05-23:
I tried different read intervals. First it was 5 seconds, later I tried 15 seconds and 1 minute also.
For writing I always have a 1 minute interval, which was working without problems.
By the way it is the same project, for which I started case #641
Can the updates in the meantime have had an effect on the behaviour? Someone else (see #690) seems to have similar problems like me in the last weeks.

TobiasKorn, 2024-05-27:
In the meantime I found out, that it seems to be a network problem. Because I read a structure the most time is used for browsing the subnodes of the structure and then reading their values. However, the struct contains only 11 values (5 bool, 5 double integer, 1 real), but when I trace communication with wireshark it takes hundreds of request until the whole structure is read out (after first connect). Is this behaviour normal?

[image]
(after about line 1200 in the logfile I get the response in NodeRed)

mikakaraila, 2024-05-28:
Hmm, seems to be some low level problem that I cannot comment. If you need more detailed explanation you should ask Etienne (node-opcua), but it will need support contract...

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila