mikakaraila/node-red-contrib-opcua #700: How to create a table with different variables from a OPC UA server.

nicoleu23, 2024-05-23:
Hello everyone,

I am new on here and debutant in Node Red. I am trying to create a table with the dashboard template "Table" but it's without success.

I am using the OPC UA client component node and connecting it  my PLC data variables, I would like to create a table with many rows and columns corresponding to my program variables to visualize the output. I am trying but I am only getting data of one column.

Would you please direct and guide me on how to pair a table and opc ua item and client read.

Thanks,

Nicole

mikakaraila, 2024-05-28:
Have you looked examples and OPCUA-TEST-NODES.json?

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila