mikakaraila/node-red-contrib-opcua #701: Can't catch error: "Write multiple as array session is not active" and then reconnect

regrog, 2024-05-29:
This is my WRITE MULTIPLE flow.
[image]

The machine I'm connecting to is being turned off during night, but in the morning when I try to send a write request I'm having this error:
> "Client node error on: W 1 error: ["Write multiple as array session is not active!"]"

Then I must set the Client-node Endpoint to opc.tcp://0.0.0.0:4840/, deploy and then reset to the correct IP.

Request to the endpoint `test-write`:
```
[
    {
        "nodeId": "ns=4;s=GVL.stOpcIn.QUANTITY2",
        "datatype": "UInt64",
        "value": 580
    }
]
```

My problem is that I can't catch the error because both status and catch nodes don't receive the error so I can't retry a connection.
Is there a way to retry connection to the server if the session is not active?
If possible I'd prefer to reconnect during the API request avoiding it to fail.

this is my flow json:  [write.json](https://github.com/mikakaraila/node-red-contrib-opcua/files/15482064/write.json)

Thanks!

mikakaraila, 2024-05-29:
datatype UInt64 contains 2 bytes...  try with some other type first.
Value must be given another way.

regrog, 2024-05-29:
The problem is not the type datatype, with a String is the same.

Anyway I made a mistake in the flow due to many changes to find a solution... the debug node "D Status 1" is disabled here, so I have to rectify what I wrote.
I can read the msg "invalid session" in the Status node and perform a connection but the main API request will wait until timeout.
Is there a way to easily reconnect and to carry on the main write request?

closed 2024-10-14 by mikakaraila