mikakaraila/node-red-contrib-opcua #702: Better OPC-Client Description

muratcanberber, 2024-05-30:
I have rewrited OPC-UA Client description from ChatGPT. Would you consider to use it ?

Here's a revised and clearer version of the document, with added code examples for better understanding:

---

# OpcUa-Client Guide

## Connecting to an Endpoint
To connect to an OPC UA endpoint, use the following format:
```
opc.tcp://host:port/UA/EndpointName
```

## Actions

### Read
Reads the value of a node.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
return msg;
```

### Write
Writes a value to a node.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
msg.payload = 42;  // Value to write
return msg;
```

### Browse
Browses the address space.
```javascript
msg.topic = "ns=2;s=Channel1.Device1";
return msg;
```

### Subscribe
Subscribes to changes of a node value.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
msg.payload = {
    interval: 100  // Interval in ms
};
return msg;
```

### Unsubscribe
Unsubscribes from a node.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
return msg;
```

### Delete Subscription
Deletes a subscription.
```javascript
msg.topic = "delete_subscription";
return msg;
```

### Event
Subscribes to events.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Event";
return msg;
```

### Info
Reads attributes from a node.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
return msg;
```

### Build
Constructs an Extension Object message.
```javascript
msg.topic = "build";
msg.datatype = "MyDataType";
msg.payload = {
    field1: "value1",
    field2: "value2"
};
return msg;
```

### Monitor
Monitors a node with additional parameters.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
msg.payload = {
    interval: 100,  // Interval in ms
    deadbandType: "Absolute",
    deadbandValue: 0.1
};
return msg;
```

### Read Multiple
Reads multiple nodes.
```javascript
msg.payload = ["ns=2;s=Channel1.Device1.Tag1", "ns=2;s=Channel1.Device1.Tag2"];
msg.topic = "read_multiple";
return msg;
```

### Write Multiple
Writes values to multiple nodes.
```javascript
msg.payload = [
    {nodeId: "ns=2;s=Channel1.Device1.Tag1", value: 42},
    {nodeId: "ns=2;s=Channel1.Device1.Tag2", value: 43}
];
msg.topic = "write_multiple";
return msg;
```

### Register
Registers node IDs.
```javascript
msg.payload = ["ns=2;s=Channel1.Device1.Tag1"];
msg.topic = "register";
return msg;
```

### Unregister
Unregisters node IDs.
```javascript
msg.payload = ["ns=2;s=Channel1.Device1.Tag1"];
msg.topic = "unregister";
return msg;
```

### Acknowledge
Acknowledges an event/condition.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Alarm";
msg.conditionId = "ns=2;s=Channel1.Device1.Event";
msg.comment = "Acknowledged";
return msg;
```

### History
Reads historical values or aggregates.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Tag1";
msg.aggregate = "raw";
msg.start = new Date(Date.now() - 3600000);  // 1 hour ago
msg.end = new Date();
return msg;
```

### Read File
Reads a file from the server.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.File";
return msg;
```

### Write File
Writes a file to the server.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.File";
msg.payload = Buffer.from("File content");
return msg;
```

### Connect
Connects to the server.
```javascript
msg.topic = "connect";
return msg;
```

### Disconnect
Disconnects from the server.
```javascript
msg.topic = "disconnect";
return msg;
```

### Reconnect
Reconnects to the server.
```javascript
msg.topic = "reconnect";
return msg;
```

### Method
Calls a method on the server.
```javascript
msg.topic = "ns=2;s=Channel1.Device1.Method";
msg.payload = {
    inputArguments: [42, "arg2"]
};
return msg;
```

## Server Certificate Options

- **None**: No certificate required.
- **Local certificate file (.pem)**: Path to local certificate file.
- **Local private key file (.pem)**: Path to local private key file.

The client will try to load the server certificate from the endpoint if available. Manually copying the server certificate is also an option. Use a relative path and filename for the server certificate (must have a .pem extension).

## Inject Node Usage
Inject your OPC UA address (NodeId) by the Topic of an Inject node or with the OpcUa-Item controlled by an Inject node.

## Status Monitoring
The second output of the node provides status information:
```javascript
{
    error: "error message",
    endpoint: "opcuaEndpoint",
    status: "currentStatus"
}
```

For more detailed examples, refer to the examples folder and the flow file `OPCUA-MULTI-SUB.json`.

---

mikakaraila, 2024-05-31:
For me it does not look accurate, missing action!

integratec, 2024-06-17:
First look... also missing datatype for action write.