mikakaraila/node-red-contrib-opcua #703: BadNodeIdUnknown (0x80340000) The node id refers to a node that does not exist in the server address space

mrajar1982, 2024-06-05:
Hello @mikakaraila ,
Am a new user in node red, recently created a flow getting the data from OPC UA sending to MQTT, while doing this am getting error like ""StatusCode: BadNodeIdUnknown (0x80340000) The node id refers to a node that does not exist in the server address space.", the OPC UA tag is "ns=2;s=1957.SPRO.Production_Part_Number" but while using the same flow with some other tag "ns=2;s=_System._Time_Hour24", it working good, non other than "ns=2;s=_System._Time_Hour24" tags are not working throwing the same error.
Could you help on that - Flow details
 
[flows (1).json](https://github.com/user-attachments/files/15595558/flows.1.json)
[flows (1).json](https://github.com/user-attachments/files/15595558/flows.1.json)
[flows (1).json](https://github.com/user-attachments/files/15595558/flows.1.json)

mikakaraila, 2024-06-06:
Check nodeId with UaExpert, there is most probably some type in the string.

mrajar1982, 2024-06-06:
Ok, let me try thanks for your response

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila