mikakaraila/node-red-contrib-opcua #707: Read/Write ExtensionObject (data type) between NX102 & PLCnext using OPC UA & REST api.

Maxime771, 2024-06-18:
Hi, this is how I try to establish read/write communication for a structure variable between an NX102 & PLCnext.

I am using OPC UA protocol to read a data structure from the NX102 as shown below : 
My DataType that I want to read/write : 
[image]
In UaExpert, we can see the following information : 
 [image]
In Node-Red, I can read  '' staaMCtest '' and its members from the NX102.
[image]
Using the fonction "Map to PLCnext", I can write each member of staaMCtest to the PLCnext structure  :
[image]
In the PLCnext, we can see that the value are correctly written in ''teststruc'' of type MyStruct :
[image]
I set the variable teststruc as a HMI tag, so it can use the REST data interface.
[image]

Since the value of staaMCtest (NX102) and teststruc (PLCnext) can be change in both PLC, I want to send the value of teststruc back to staaMCtest (NX102) :
[image]
We can see that the content in the msg payload of ''debug 6" is the same as the content in "debug 1" (showed in previous image)

My issue is writing via the function bloc "Omron NX102 Client Write". I got this error : 
[image]

Any ideas how can we solve this issue?

Maxime771, 2024-06-18:
Were the Node-RedFlow : 
[
    {
        "id": "b0d5c20d2c81e15a",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "49cca7a35053b110",
        "type": "OpcUa-Client",
        "z": "b0d5c20d2c81e15a",
        "endpoint": "40fb7f127b35e0f0",
        "action": "read",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "useTransport": false,
        "maxChunkCount": 1,
        "maxMessageSize": 8192,
        "receiveBufferSize": 8192,
        "sendBufferSize": 8192,
        "name": "Omron NX102 Client Read",
        "x": 200,
        "y": 180,
        "wires": [
            [
                "2e8343ce1a45409c",
                "f0a7e671e6104770"
            ],
            []
        ]
    },
    {
        "id": "2e8343ce1a45409c",
        "type": "debug",
        "z": "b0d5c20d2c81e15a",
        "name": "debug 1",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 460,
        "y": 160,
        "wires": []
    },
    {
        "id": "5c269e11d32f8ade",
        "type": "inject",
        "z": "b0d5c20d2c81e15a",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
[... 220 lines cut ...]
        "y": 480,
        "wires": []
    },
    {
        "id": "40fb7f127b35e0f0",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.78.170:4840",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    },
    {
        "id": "e6ec980a1af28a84",
        "type": "plc-connector",
        "host": "192.168.78.169",
        "station_name": "axcf2152-pnc",
        "session_timeout": "20000",
        "api_version": "v1.4",
        "auth_required": "false"
    }
]

mikakaraila, 2024-06-20:
Hmm, it could be that a value is checked on PLC and it will reject value with statuscode out of range.
String to long or some value is smaller/bigger than it will accept...

Maxime771, 2024-06-20:
Could it be because Node-Red don't recognize the type of variable aaMC_testStructure?
[image]
Strange... because UaExpert see aaMC_testStructure as a ExtensionObject... (like shown in previous images)

Maxime771, 2024-06-20:
Nvm I tried an other OPC UA browser and aaMC_testStructure data type is ExtensionObject :
[image]

Maxime771, 2024-06-20:
Im even just trying to write some values in aaMC_testStructure and I get BadNodeIdUnknow error..
[image]
[image]
If I change datatype to ''ExtensionObject'' in the fonction staaMCtest I get the same BadOutOfRange error.