mikakaraila/node-red-contrib-opcua #709: On msg.action : Disconnect to a OPC UA Client the Node Red Application crashes

edcWre, 2024-06-27:
Hi, 
I am using node-red-contrib-opc-ua to fetch data from a Sinumerik 840 dsl. Unfortunately, I have the problem that my Node-RED application crashes when the Sinumerik has been off overnight.
Is there a way to automatically disconnect the connection at the end of the workday and reconnect it in the morning? I tried using msg.action: disconnect from the example folder, but the application crashes as well.

 Thank you very much for your help.

In the following im going to show you the error Message im getting shortly after i inject the disconnect: 

[image]

and a picture of my flow : 
[image]

my system im Using : 
27 Jun 12:57:56 - [info] Node-RED version: v3.1.9
27 Jun 12:57:56 - [info] Node.js  version: v20.13.1
27 Jun 12:57:56 - [info] Windows_NT 10.0.22631 x64 LE
27 Jun 12:57:57 - [info] Loading palette nodes
27 Jun 12:57:57 - [info] Dashboard version 3.6.5 started at /ui

node-red-contrib-opc ua : 
0.2.329

mikakaraila, 2024-07-01:
Hmm, most probably debug warning seems to use parameter that is not defined and tries to show name of the node.

edcWre, 2024-07-01:
Okay thank you for your thougths . 
Is there a way to do a similar action then Disconnect with something else ?

edcWre, 2024-07-02:
Another Question that i have. Im using node red in a factory and if the maschines are turned off the clinet trys to reconnect and mostly on the next morning the application has crashed. Is there a solution for this problem ?

mikakaraila, 2024-07-03:
I think I should update node-opcua package. The latest version contains a lot improvements for edge / network connection issues.

mikakaraila, 2024-09-10:
I added check that should prevent crash.

closed 2024-09-10 by mikakaraila