mikakaraila/node-red-contrib-opcua #712: Issue With Data Change Filter in 'MONITOR'

ParkMartins, 2024-07-03:
When using the 'MONITOR' action in the 'OpcUa-Client' node of the _node-red-contrib-opcua_ package, I have encountered an issue with the configuration of the deadband in 'Absolute' or 'Percentage' mode. Upon initial deployment, both deadband configurations function as expected for the first subscription. However, subsequent subscriptions do not maintain this functionality, leading to the deadbands becoming inactive for all monitored value changes.

I have tried to define the configuration directly in the 'Opc Ua-Client' node and also on a function node. Both implementations work exactly the same way, like described above. 

[image]

**Expected Behavior:** The 'Absolute' and 'Percentage' deadband configurations should consistently remain active for every monitored item. This is crucial for reliable data change monitoring.

mikakaraila, 2024-07-03:
Hmm, I expect root cause could be without seeing your address space that variable should contain EURange that it could calculate from the min & max the percentage.

I need example flow to check & repeat problem, can you provide it?

ParkMartins, 2024-07-09:
Thanks for your answer, here is the flow so you can analyse it in detail. 

[DataChangeFilter Nodes.json](https://github.com/user-attachments/files/16142150/DataChangeFilter.Nodes.json)

For your information, I have also tested the behaviour in parallel with UaExpert and it works fine, so the problem can't be related to the variable being read from the server. In the printscreens bellow you can see this behaviour with a configured absolute value change of 1. In this example, with value = 111, the monitoring funtion should only change when I the absolute is > 1, which is not happening. I have the exact same configuration in UaExpert and the behaviour there is correct. The server value that is being forced is the left picture on the second image.

[image]

[image]

mikakaraila, 2024-09-10:
Inject empty string instead of timestamp. Timestamp in the payload is used for publish interval...

ParkMartins, 2024-10-18:
Thanks a lot, this was actually the cause of the problem. I can confirm that the data change filter is now working.

closed 2024-10-18 by ParkMartins