mikakaraila/node-red-contrib-opcua #717: Issue when using version 0.2.331

jamalsal, 2024-07-24:
Hi. I encountered the following error when using the OPC UA Client Node on version 0.2.331:
"StatusCode: BadNotReadable (0x803a0000) The access level does not allow reading or subscribing to the Node."
When downgrading to version 0.2.310, it was possible to establish a connection with the server, without any modifications to the node.

RudolfTheRedNode, 2024-07-26:
Hi @mikakaraila, last previous version got my program working in multiple write,. now whenever I trigger this I got errors:

[31mClient node error on: AMS OPC Client WRITE2 error: ["Write multiple as array session is not active!"][39m

This is my function code below:

msg.payload = [];

msg.topic = "write";

msg.payload[0] =
{
    nodeId: "ns=3;i=16847616",
    datatype: "Boolean",
    value: true
}

msg.payload[1] =
{
    nodeId: "ns=3;i=16847104",
    datatype: "Boolean",
    value: true
}

return msg;

mikakaraila, 2024-08-27:
To write values msg.topic should be "writemultiple"