mikakaraila/node-red-contrib-opcua #719: Error when trying to unsubscribe

Poeschie, 2024-08-02:
Hello,

i get an error when i'm trying to unsubscribe from a variable on my OPC UA Server. See attached a screenshot of the debug messages.

First I subscribe to a variable which is changing frequently. Then i get values of the variable in entered interval.

But when i try to unsubscribe from this variable i get an error. Where is my error?

[image]

My Flow looks like this:
`[
    {
        "id": "912800e7.e00308",
        "type": "tab",
        "label": "Flow 1"
    },
    {
        "id": "ed6442e94f77eb97",
        "type": "OpcUa-Client",
        "z": "912800e7.e00308",
        "endpoint": "f6af4a4f7d984a0d",
        "action": "subscribe",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": "5",
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "useTransport": false,
        "maxChunkCount": "",
        "maxMessageSize": "",
        "receiveBufferSize": "",
        "sendBufferSize": "",
        "name": "",
        "x": 780,
        "y": 280,
        "wires": [
            [
                "e083bbcaaff38a8a"
            ],
            []
        ]
    },
    {
        "id": "e083bbcaaff38a8a",
        "type": "debug",
        "z": "912800e7.e00308",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 950,
        "y": 280,
        "wires": []
    },
    {
        "id": "b2b8559e9d1473ad",
        "type": "OpcUa-Item",
        "z": "912800e7.e00308",
        "item": "ns=2;s=AL-Master.AE4",
        "datatype": "Double",
        "value": "",
        "name": "",
        "x": 400,
        "y": 280,
        "wires": [
            [
                "05dc4309d53cfc2b"
            ]
        ]
    },
    {
        "id": "25e00916bdb50e92",
        "type": "function",
        "z": "912800e7.e00308",
        "name": "Unsubscribe",
        "func": "msg.action=\"unsubscribe\";\nreturn msg;",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 590,
        "y": 360,
        "wires": [
            [
                "ed6442e94f77eb97",
                "c9c8d3378d8d3369"
            ]
        ]
    },
    {
        "id": "85efbdaaf67949c2",
        "type": "OpcUa-Item",
        "z": "912800e7.e00308",
        "item": "ns=2;s=AL-Master.AE4",
        "datatype": "Double",
        "value": "",
        "name": "",
        "x": 400,
        "y": 360,
        "wires": [
            [
                "25e00916bdb50e92"
            ]
        ]
    },
    {
        "id": "4f86f511bd75df78",
        "type": "inject",
        "z": "912800e7.e00308",
        "name": "Test",
        "repeat": "",
        "crontab": "",
        "once": false,
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "x": 230,
        "y": 280,
        "wires": [
            [
                "b2b8559e9d1473ad"
            ]
        ]
    },
    {
        "id": "523c6a96bdea9a2b",
        "type": "inject",
        "z": "912800e7.e00308",
        "name": "Test",
        "repeat": "",
        "crontab": "",
        "once": false,
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "x": 230,
        "y": 360,
        "wires": [
            [
                "85efbdaaf67949c2"
            ]
        ]
    },
    {
        "id": "05dc4309d53cfc2b",
        "type": "function",
        "z": "912800e7.e00308",
        "name": "Subscribe",
        "func": "msg.action=\"subscribe\";\nreturn msg;",
        "outputs": 1,
        "noerr": 0,
        "x": 580,
        "y": 280,
        "wires": [
            [
                "ed6442e94f77eb97",
                "3e6f2d548edde3c3"
            ]
        ]
    },
    {
        "id": "c9c8d3378d8d3369",
        "type": "debug",
        "z": "912800e7.e00308",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 950,
        "y": 360,
        "wires": []
    },
    {
        "id": "3e6f2d548edde3c3",
        "type": "debug",
        "z": "912800e7.e00308",
        "name": "",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "true",
        "targetType": "full",
        "statusVal": "",
        "statusType": "auto",
        "x": 750,
        "y": 180,
        "wires": []
    },
    {
        "id": "f6af4a4f7d984a0d",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://172.17.4.157:52530/airleader",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

aq-daniel, 2024-08-07:
Hi,
I face the same problem with unsubscribing. The error message "Cannot unscubscribe, no subscription" appears, but the subscription remains active and new value changes are returned for the subscribed nodes.
I also tried the flow mentioned [here](https://discourse.nodered.org/t/how-to-unsubscribe-opc-ua-nodes/69130). Unsubscribing doesn't work despite using the subscriptionId or using "currentSession" with action "deletesubscription".

Is it worth to try an older release of this package? Or do you have any other ideas for a quick fix? Thanks.

I'm using the latest dockerized NodeRed v4.0.2 with node-red-contrib-opcua v0.2.331

aq-daniel, 2024-08-09:
I've installed package version 0.2.330 and unsubscribing seems to work just fine with the previous package version.

Poeschie, 2024-08-13:
Yes for me the same. In Version 0.2.330 i'm able to unsubscribe with the same flow. I will go with this version until it is fixed.
Thank You for this hint.

mikakaraila, 2024-08-27:
I will try to find some to look this one... something changed that I am not aware.

hopinll, 2024-08-28:
[image]

It seems that there is an incorrect variable scope.

mikakaraila, 2024-09-10:
Did you look my example OPCUA-TEST-NODES.json that you have to store topic that is subscriptionId to unsubscribe.
If you want to terminate specific nodeId it is terminate that you want to do perhaps...

MountainKing91, 2024-09-11:
Following this, as I am experiencing the same after upgrading to v0.2.331.

@mikakaraila When subscribing I store the subscritpionId and I am passing that as msg.topic when I unsubscribe, no errors or warning but still it does not unsubscribe.

Also, deletesubscription seems not to work too (I am actually subscribed):
[image]

mikakaraila, 2024-09-11:
Hmm, I just checked code and there was local variable. Fixed code.

closed 2024-10-14 by mikakaraila