mikakaraila/node-red-contrib-opcua #720: Server is not running

raulus127, 2024-08-02:
I am trying to run an opc ua server with this json that I attached. It doesn't work for me. I've seen other threads on this forum but it seems like something else is happening. The error that appears is "Server node error on:  error: "Server is not running". Can you help me please? Thank you so much!

[pp.json](https://github.com/user-attachments/files/16472591/pp.json)

The detail of the error:

Welcome to Node-RED
===================

2 Aug 11:35:41 - [info] Node-RED version: v4.0.2
2 Aug 11:35:41 - [info] Node.js  version: v20.16.0
2 Aug 11:35:41 - [info] Windows_NT 10.0.22631 x64 LE
2 Aug 11:35:41 - [info] Loading palette nodes
using nodejs crypto (native)
+-------------------------------------------------------------------------------------+
| Warning:                                                                            |
| node-opcua-client-crawler module has been deprecated and is not maintained anymore. |
| Please use '@sterfive/crawler' instead.                                             |
| '@sterfive/crawler' is available to the NodeOPCUA Subscription members              |
+-------------------------------------------------------------------------------------+
2 Aug 11:35:44 - [info] Dashboard version 3.6.5 started at /ui
14:35:44.409Z :verify_pcks1                  :54    [NODE-OPCUA-W27] node version v20.16.0
14:35:44.410Z :verify_pcks1                  :55      you need to use node flag --security-revert=CVE-2023-46809 if you have issue with RSA PKCS#1 v1.5
2 Aug 11:35:44 - [info] Settings file  : C:\Users\raulm\.node-red\settings.js
2 Aug 11:35:44 - [info] Context store  : 'default' [module=memory]
2 Aug 11:35:44 - [info] User directory : C:\Users\raulm\.node-red
2 Aug 11:35:44 - [warn] Projects disabled : editorTheme.projects.enabled=false
2 Aug 11:35:44 - [info] Flows file     : D:\Acindar\Tareas\Balanza 5\OPC UA\pp.json
2 Aug 11:35:44 - [warn]

---------------------------------------------------------------------
Your flow credentials file is encrypted using a system-generated key.

If the system-generated key is lost for any reason, your credentials
file will not be recoverable, you will have to delete it and re-enter
your credentials.

You should set your own key using the 'credentialSecret' option in
your settings file. Node-RED will then re-encrypt your credentials
file using your chosen key the next time you deploy a change.
---------------------------------------------------------------------

2 Aug 11:35:44 - [warn] Encrypted credentials not found
2 Aug 11:35:44 - [info] Server now running at http://127.0.0.1:1880/
2 Aug 11:35:44 - [info] Starting flows
2 Aug 11:35:44 - [info] Started flows
2 Aug 11:35:45 - [error] [OpcUa-Server:8678134d0d55c7e5] Server node error on:  error: "Server is not running"
2 Aug 11:35:45 - [error] [OpcUa-Server:8678134d0d55c7e5] Server node error on:  error: "Server is not running"
2 Aug 11:35:45 - [error] [OpcUa-Server:8678134d0d55c7e5] Server node error on:  error: "Server is not running"
14:35:45.680Z :message_builder_base          :144   Error   client1 Error message not specified
14:35:45.682Z :register_server_manager       :526   RegisterServer to the LDS  has failed during secure connection  => please check that you server certificate is trusted by the LDS.
 ...                                                err: The connection may have been rejected by server,
 ...                                                Err = (ClientTCP_transport1: socket has been disconnected by third party)
 ...                                                LDS endpoint    : opc.tcp://PCRAUL:4840
 ...                                                security mode   : SignAndEncrypt
 ...                                                security policy : http://opcfoundation.org/UA/SecurityPolicy#Basic256
2 Aug 11:35:46 - [warn] [OpcUa-Server:8678134d0d55c7e5] Mandatory parameter 'namespace' is missing
2 Aug 11:35:46 - [warn] [OpcUa-Server:8678134d0d55c7e5] warning: properties like messageType, namespace, variableName or VariableValue is missing.
2 Aug 11:35:46 - [warn] [OpcUa-Server:8678134d0d55c7e5] Mandatory parameter 'namespace' is missing
2 Aug 11:35:46 - [warn] [OpcUa-Server:8678134d0d55c7e5] warning: properties like messageType, namespace, variableName or VariableValue is missing.
2 Aug 11:35:46 - [error] [OpcUa-Server:8678134d0d55c7e5] TypeError: Cannot read properties of undefined (reading 'hasOwnProperty')
    at contains_messageType (C:\Users\raulm\.node-red\node_modules\node-red-contrib-opcua\opcua\104-opcuaserver.js:584:28)
    at contains_necessaryProperties (C:\Users\raulm\.node-red\node_modules\node-red-contrib-opcua\opcua\104-opcuaserver.js:605:17)
    at OpcUaServerNode._inputCallback (C:\Users\raulm\.node-red\node_modules\node-red-contrib-opcua\opcua\104-opcuaserver.js:540:17)
    at C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:214:26
    at Object.trigger (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\util\lib\hooks.js:166:13)
    at Node._emitInput (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:206:11)
    at Node.emit (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:190:25)
    at Node.receive (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\nodes\Node.js:499:10)
    at deliverMessageToDestination (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\flows\Flow.js:803:40)
    at Immediate._onImmediate (C:\Users\raulm\AppData\Roaming\npm\node_modules\node-red\node_modules\@node-red\runtime\lib\flows\Flow.js:819:21)

raulus127, 2024-08-05:
I read in another thread that this error was solved by updating the version to v0.2.112. Will I have to do it or do I already have it updated? I downloaded the latest version from the page.

mikakaraila, 2024-09-12:
I expect you have some other troubles on your environment and most probably also inside the flow.
Can you attach flow file?

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila