mikakaraila/node-red-contrib-opcua #723: Write multiple no longer works with 0.2.331

Eldarion85, 2024-08-24:
After upgrading to 0.2.331, Write multiple no longer works. With every input I get the message "Write multiple as array session is not active!", which unfortunately doesn't help me.

After downgrading to 0.2.330, it works again straight away.

[image]

mikakaraila, 2024-08-27:
Can you attach you flow or simplified one?
I would check it and use it to debug my nodes (action).

mikakaraila, 2024-08-27:
To write values msg.topic should be "writemultiple"

Eldarion85, 2024-10-01:
Hello Mika,
I was on vacation for three weeks and just read your message. I will upgrade to the latest version later and change my topic from "multiple" to "writemultiple". If that doesn't change anything, I'll send you a sample flow.

Eldarion85, 2024-10-01:
Unfortunately, it still doesn't work after I changed the topic to "writemultiple".
However, I have a different message in version 0.2.332 than in the last test with version 0.2.331.
Instead of "Write multiple as array session is not active!" I now get the information in the debug and under the node that there are no items to write. In the screenshot you can see how all my write multiple arrays are structured that go to the OpcUa client node. The action in the node is set to "WRITE MULTIPLE".
It can't be the certificates, because the other client nodes with "Read", "Write" and "Subscribe" also run with the new version without any problems with the same certificates and even after a downgrade the "Write Multiple" runs again with the same certificates.
What else could be the reason that it works again immediately with the old version and this message appears with the new one?

[image]

[image]

[image]

Eldarion85, 2024-10-01:
I saw that in the new version you register the subscription individually on the server. Here is a screenshot from the Siemens PLC. Before, there was only one entry (yellow frame) with the total number of all variables (173). Now I can also see the individual subscriptions in the server (green frame), which is a very good improvement.
Thank you very much for that!
Now I just have to solve the problem with Writemultiple :).

[image]

apbrauer73, 2024-10-09:
@mikakaraila How did this get resolved?  I have the same issue, if I downgrade to 0.2.330 it works.  I upgraded to version 0.2.332.  Then based on this thread I changed my topic to "write multiple", but now I get the error "Client node error on: Kepware Write error: ["Kepware Write no items to write"]".   Not sure why the OpcUa client node reports no items to write when the debug message shows items in the payload.  What I am I missing?

[image]

mikakaraila, 2024-10-10:
I find time to look this and some stupid error done by me (merge or other fault).
Got this fixed easily...

apbrauer73, 2024-10-11:
@mikakaraila I tried the latest version 0.2.333.  I restarted Node Red after the updating the "node-red-contrib-opcua" palette. I triggered my flow and it failed again for the same error.  

[image]

Eldarion85, 2024-10-11:
Thanks for the test, I'll stick with the old version for now.
I saw today that a new version was released and wanted to test it tomorrow as well.

closed 2024-10-14 by mikakaraila