mikakaraila/node-red-contrib-opcua #725: Subscribe to the OPC UA client

joliveira82, 2024-08-28:
Hello, I'm having a problem subscribing to OPC UA Block
as shown in the picture I'm subscribing to variables and I want to do the same for another IP, but as you can see in the lower right corner of the image, there is the debug where it shows that it returns either the value I want which is true or false and it also returns a payload with a random value with the topic is not any of the names that are next to it.

Can you help me? 

I already had this same code and it worked, but after updating it stopped working.

[image]

mikakaraila, 2024-08-29:
Change debug node to show whole msg object...

joliveira82, 2024-08-30:
Now node-red isn't showing up when I subscribe. it just shows this.
[image]

But if I put it in read, it already appears like this and that's how it used to appear when it was all working.And I wanted it to always be like this, as it worked before.
[image]

mikakaraila, 2024-09-02:
Try to inject empty string instead of timestamp. If I remember correctly it will be used in some cases as publish interval...

joliveira82, 2024-09-03:
I think that's what you mean. If it is, it keeps going wrong and I don't understand what it could be.
[image]

joliveira82, 2024-09-03:
I've been testing and this time I've put the name of a variable directly into the inject block and with the opc ua client block in subscription action every 1s and as you can see in the direct debug column it was switching between giving and not giving until it stopped giving.
[image]

mikakaraila, 2024-09-03:
Inject only once, if you repeat then server will be out of resources (amount of subscritions).

joliveira82, 2024-09-03:
when I click only once on the inject block, this appears.
[image]

mikakaraila, 2024-09-03:
Correct output, subscriptionId can be used to terminate subscription and as value changes then topic is nodeId and payload is value.

joliveira82, 2024-09-03:
does this mean that it's normal for that large value to appear before the value of the variable?

[image]

I didn't want these values to appear. Is there any way of eliminating them?

mikakaraila, 2024-09-03:
Compare msg.topic != "subscriptionId" all others are with msg.topic = nodeId + msg.payload is actual value

joliveira82, 2024-09-03:
Thanks now, you've solved that problem.

Now I have another problem where I don't read all the variables. I've separated the variables into different function blocks so that they're more organized, but I'm having the problem that node red doesn't read them all, it only reads the ones with the fewest variables and it only does this when the injection cycle is shorter than that of the function with the most variables.
At the moment it has 3 inject blocks, but it used to have only one and it worked the same way.

[image]