mikakaraila/node-red-contrib-opcua #727: Use FlowFuse to install node-red-contrib-opcua and Configure as shown in the picture， [error]   Invalid endpoint

dingling1990, 2024-09-09:
Look forward to your reply

[image]

[image]

2024/9/9 10:23:14  []        :         using securityPolicy: ["None"]
2024/9/9 10:23:14  []        : Case B: UserName & password does not match to server (needed by Sign or SignAndEncrypt), check username: undefined and password: undefined
2024/9/9 10:23:14  []        : Case C: With Sign you cannot use SecurityPolicy None!!
2024/9/9 10:23:14  [warn]    Case C: With Sign you cannot use SecurityPolicy None!!
2024/9/9 10:23:14  [error]   Client node error on:  error: ["Wrong endpoint parameters: {\"id\":\"b8dfea0ac9e4dba2\",\"type\":\"OpcUa-Endpoint\",\"_closeCallbacks\":[],\"_inputCallback\":null,\"_inputCallbacks\":null,\"_expectedDoneCount\":0,\"wires\":[],\"_wireCount\":0,\"credentials\":{},\"endpoint\":\"opc.tcp://10.30.136.3:49310\",\"securityPolicy\":\"None\",\"securityMode\":\"None\",\"login\":false,\"none\":true,\"usercert\":false,\"userCertificate\":\"\",\"userPrivatekey\":\"\",\"user\":null,\"password\":null}"]
2024/9/9 10:23:14  [error]   Invalid endpoint

dingling1990, 2024-09-09:
debug view

[image]

mikakaraila, 2024-09-09:
It could be that openssl installation (download) has failed on the background... it is used to generate certificates at installation phase even those are not used.