mikakaraila/node-red-contrib-opcua #728: UInt64 support in opcua server

jhooney, 2024-09-09:
UInt64 support was added to the opcuaread and opcuawrite nodes in #98 using the format [highByte, lowByte]. The opcuaitem node responds with [UInt64, NaN] and the opcuaserver node does not support UInt64 at all. My application uses a UInt64 bit number to track a unix timestamp in ms, and will not work without UInt64. I think I see a way to add this functionality, so I will try to submit a PR shortly.

Here is a test flow I made to look at the issue: 
[UInt64TestFlows.json](https://github.com/user-attachments/files/16937600/UInt64TestFlows.json)

mikakaraila, 2024-09-10:
I will publish new version later today

closed 2024-09-10 by mikakaraila

jhooney, 2024-09-11:
Thanks for being so speedy!