mikakaraila/node-red-contrib-opcua #731: Cannot properly set browseName for a folder in the server node

jhooney, 2024-09-11:
I am trying to create a folder with a browseName that is different from its nodeId to match the structure of a server that I have running on a plc (running on codesys on the RIO from Opto22). 
There seems to be two intended ways to set a browseName of a node, either send a message in the form a: 
```
{
    payload: { "opcuaCommand": "addFolder" },
    topic: "ns=4;s=Sub_Folder1;browseName=Sub1"
}
```
or b: 
```
var msg2 = {
    payload: { "opcuaCommand": "addFolder" },
    topic: "ns=4;s=Sub_Folder2;b=Sub2"
}
```

The behavior I am looking for is that the NodeID becomes `ns=4;s=<folder-name>` and the browseName becomes the browseName component. If I use method a the nodeId becomes ns=4;s=Sub_Folder1, but the node's browseName does not change, and if I use method b the nodeId becomes ns=4;s=Sub_Folder2;b=Sub2, but the browseName does change successfully. Here is what each folder looks like when viewed in UaExpert: 
[image]
[image]

Both methods get close to the functionality I'm looking for but seem to not fully implement a browseName. I'm not sure if this is a bug, or if there is another way to set browseName that I am missing. Seems like it is probably a simple fix, if I get some free time I'll try to submit a PR but pretty booked up for awhile.

Here is a flow I made to demonstrate: 
[setFolderBrowseNameFlow.json](https://github.com/user-attachments/files/16965234/setFolderBrowseNameFlow.json)

mikakaraila, 2024-09-12:
In my code there is format example, but I will study why it does not set browseName correctly... your test case looks OK.

// format is msg.topic="ns=1;s=Main.Test;browseName=Test"

mikakaraila, 2024-09-12:
Small change and seems to work:
[image]

johnatark, 2024-09-12:
Interesting, it seems that browseName can be set when namespace = 1 but not when namespace = 4. I did a little more testing, looks like it also fails when namespace is 2 or 3.

jhooney, 2024-09-12:
Interesting, it seems that browseName can be set when namespace = 1 but not when namespace = 4. I did a little more testing, looks like it also fails when namespace is 2 or 3.

mikakaraila, 2024-09-13:
Thanks, I will look and make a fix...

closed 2024-10-14 by mikakaraila