mikakaraila/node-red-contrib-opcua #734: CreateSession --> LocaleIds Parameter

robotsox, 2024-09-30:
Hi,

can you please add the "LocaleIds" Property to change the displayed textLocale from the alarm-events?
my problem is, that i cannot change the displayed text language from my plc-alarms to German. The opc server from the plc requiered the "LocaleIds" Property to switch from English to German.

[image]
[image]

mikakaraila, 2024-09-30:
This will actually require support at the node-opcua level...

I tested this and asked Etienne, but he asks for some funding/sponsor to add this feature.

robotsox, 2024-09-30:
What should the financing/sponsoring look like? :)

mikakaraila, 2024-09-30:
Better to ask directly from Etienne (Etienne Rossignon <etienne.rossignon@sterfive.com>).
He will give some estimate / alternatives...

robotsox, 2024-09-30:
i have a reply on the node-opcua issue. 
is there a similar easy integration in node-red-opc-ua? :)

https://github.com/node-opcua/node-opcua/issues/1392#issuecomment-2383222581

[image]

erossignon, 2025-01-27:
> Better to ask directly from Etienne (Etienne Rossignon [etienne.rossignon@sterfive.com](mailto:etienne.rossignon@sterfive.com)). He will give some estimate / alternatives...

@robotsox : you (or your company) can become a backer by applying at the [node-opcua open-collective](https://opencollective.com/node-opcua/contribute/backer-23290/checkout)