mikakaraila/node-red-contrib-opcua #736: passing the ServerIP over msg.-Object

robotsox, 2024-10-03:
Can you implate that we can send over the msg.-object the opc-ua server ip/host and the port number?
Thanks!

mikakaraila, 2024-10-03:
It's already implemented for the client.
You can specify endpoint parameters

robotsox, 2024-10-03:
ok, how it works? which message object is it?

robotsox, 2024-10-04:
[image]

i tried to send a object "msg.endpoint" with the value ""opc.tcp://172.30.0.1:48400" but the node "OpcUa-Client" ignore it and use the preconfigured connection and not the dynamic value from the message object.

mikakaraila, 2024-10-04:
Did you look my examples? OPCUA-TEST-NODES.json

robotsox, 2024-10-07:
i tried your example flow with the following message objects:

```
msg.action = 'connect';
msg.OpcUaEndpoint = {}; // Initialize msg.OpcUaEndpoint as an empty object
msg.OpcUaEndpoint.endpoint = 'opc.tcp://H7Q8Q13.mshome.net:53530/OPCUA/SimulationServer';
msg.OpcUaEndpoint.securityMode = 'None';
msg.OpcUaEndpoint.securityPolicy = 'None';
msg.OpcUaEndpoint.login = false;
``` 

this object  i send to a OPC-UA-Client Node with "Read Multiple", but the settings from the message object are not accepted. 
The OPC-UA-CLient Node try only the Endpoint what is manually configured in the OPC-UA-CLient Settings.

mikakaraila, 2024-10-10:
There is something that is not anymore working, can be caused some refactoring/merge... investigating.

mikakaraila, 2024-10-11:
Found it!

closed 2024-10-12 by robotsox