mikakaraila/node-red-contrib-opcua #737: Change Source Timestamp

vic-951, 2024-10-08:
Hello there,

I have a problem with the source timestamp, I use OPC-UA server for historical data for another energysoftware. My problem ist that I get every few minutes cached data with the original timestamp. But everything I tried didn´t changed my sourceTimestamp on the opc-ua server.

Here is my flow, how I try to add values with timestamp, my timestamp is set in msg.timestamp. I also tried msg.sourcetimestamp, how can I change it?

[image]

`[
    {
        "id": "76c6428da6b6c5b2",
        "type": "OpcUa-Client",
        "z": "30ded7dc.266628",
        "endpoint": "1613cacca90c3f10",
        "action": "write",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "useTransport": false,
        "maxChunkCount": 1,
        "maxMessageSize": 8192,
        "receiveBufferSize": 8192,
        "sendBufferSize": 8192,
        "name": "",
        "x": 500,
        "y": 380,
        "wires": [
            [
                "5f92e478f305a75f"
            ],
            []
        ]
    },
    {
        "id": "5f92e478f305a75f",
        "type": "debug",
        "z": "30ded7dc.266628",
        "name": "debug 4",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "false",
        "statusVal": "",
        "statusType": "auto",
        "x": 660,
        "y": 380,
        "wires": []
    },
    {
        "id": "15ad9b78e5d04516",
        "type": "inject",
        "z": "30ded7dc.266628",
        "name": "",
        "props": [],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "x": 170,
        "y": 380,
        "wires": [
            [
                "91a0e91b9f236bec"
            ]
        ]
    },
    {
        "id": "91a0e91b9f236bec",
        "type": "function",
        "z": "30ded7dc.266628",
        "name": "write 10 values",
        "func": "let count = 0;\nlet maxCount = 10;\n\nfor (let i = 0; i < maxCount; i++) {\n    setTimeout(function () {\n        let msg = {\n            payload: i,\n            topic: 'ns=1;s=k_el_bea_m001_wlb',\n            timestamp: new Date('2024-10-08T14:30:00'),\n            datatype: \"Double\"\n        };\n\n        node.send(msg);\n    }, i * 1000); \n}\n\nreturn null;",
        "outputs": 1,
        "timeout": 0,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 320,
        "y": 380,
        "wires": [
            [
                "76c6428da6b6c5b2"
            ]
        ]
    },
    {
        "id": "1613cacca90c3f10",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://localhost:54840",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

[image]

mikakaraila, 2024-10-15:
Tested:
[image]

vic-951, 2024-10-15:
ok, thank you for your afford. Can you provide me some code, how you have done it?

mikakaraila, 2024-10-16:
I updated/added example into the examples folder and OPCUA-TEST-NODES.json

closed 2024-10-16 by mikakaraila