mikakaraila/node-red-contrib-opcua #740: Write Multiple no longer works

MMBud21, 2024-10-14:
Hi Mika,
Unfortunately the problem from issue #723 with WriteMultiple has not yet been resolved.

@apbrauer73 had already tested it. After you closed the post today, I also installed version 0.2.333 and tested it again. It has changed a bit, but it still doesn't work like it did in version 0.2.330.
[image]

The following behavior has changed, which makes even less sense to me than before.

If I write anything other than "writemultiple" in the topic, the values ​​arrive perfectly in the Siemens PLC. However, every time I send something, I get the error:
`"Client node error on: Write_Multiple error: ["Write multiple items session is not active!"]"`
and in the debug your client also gives the message `status: "invalid session"`.
But you can see at the very bottom of the debug that your OPC client node then gives the value 0 (successfully written) for both variables. The values ​​arrive in the PLC, but that's no use if all the error messages are written every time.
[image]

 However, if I write "writemultiple" in the topic, as you specify, the following error message appears:
`"Client node error on: Write_Multiple error: ["Write_Multiple no items to write"]"`
and therefore no more values ​​are written to the PLC. Accordingly, no feedback is received in the debug and only the message `status: "no items to write"` is displayed.
[image]

Why is it that it hasn't worked since version 0.2.330? You can see the structure of my two arrays in full in the debug, if you can identify an error. I have also tested it on several systems and even completely reinstalled Node-RED on one of them.

apbrauer73, 2024-10-14:
@MMBud21,  I decided to rewrite my flow to loop though each item to send to the OPCUA node:
[image]

I used this palette for the loop node:
[image]

This is my flow 

`[
    {
        "id": "133ddcfda9be9a1b",
        "type": "inject",
        "z": "4395b51eaabeb2f9",
        "name": "Set Alarms",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "str",
        "x": 380,
        "y": 220,
        "wires": [
            [
                "e07bca2a7b335291"
            ]
        ]
    },
    {
        "id": "e07bca2a7b335291",
        "type": "function",
        "z": "4395b51eaabeb2f9",
        "name": "Set Alarm tags",
        "func": "msg.payload = [];\nmsg.topic = \"writemultiple\";\n\nmsg.payload[0] = \n{\n    nodeId: \"ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452050_PE_FD\",\n    datatype: \"Boolean\",\n    value: false\n};\n\nmsg.payload[1] = \n{\n    nodeId: \"ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452010_PE_FD\",\n    datatype: \"Boolean\",\n    value: false\n};\n\nmsg.payload[2] =\n{\n    nodeId: \"ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U451700_PE_FD\",\n    datatype: \"Boolean\",\n    value: false\n};\n\nmsg.payload[3] =\n{\n    nodeId: \"ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452020_PE_FD\",\n    datatype: \"Boolean\",\n    value: false\n};\n\nmsg.payload[4] =\n{\n    nodeId: \"ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_DEV_ENET\",\n    datatype: \"UInt16\",\n    value: 0\n};\n\nreturn msg;",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 580,
        "y": 220,
        "wires": [
            [
                "efac4e3500ab452e",
                "06eaaa2ccd1fad5b"
            ]
        ]
    },
    {
        "id": "efac4e3500ab452e",
        "type": "debug",
        "z": "4395b51eaabeb2f9",
        "name": "Set debug",
        "active": false,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 750,
        "y": 120,
[... 129 lines cut ...]
        "reg": false,
        "x": 1160,
        "y": 320,
        "wires": [
            [
                "7f6bc79ee4a56240",
                "580867f9b12a99b6"
            ]
        ]
    },
    {
        "id": "b37371af99e005b7",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.51.127:49320",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]`

This is my workaround until this gets resolved, I hope it helps.

mikakaraila, 2024-10-15:
Ok, I will recheck what could cause this...

mikakaraila, 2024-10-15:
Found it, wrong error handler.

apbrauer73, 2024-10-15:
@mikakaraila , I just tried 0.2.334 and I still get error "no items to write".   Is this the correct payload format for the Write multiple? 

`msg.payload = [];
msg.topic = "writemultiple";

msg.payload[0] = 
{
    nodeId: "ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452050_PE_FD",
    datatype: "Boolean",
    value: false
};

msg.payload[1] = 
{
    nodeId: "ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452010_PE_FD",
    datatype: "Boolean",
    value: false
};

msg.payload[2] =
{
    nodeId: "ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U451700_PE_FD",
    datatype: "Boolean",
    value: false
};

msg.payload[3] =
{
    nodeId: "ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_U452020_PE_FD",
    datatype: "Boolean",
    value: false
};

msg.payload[4] =
{
    nodeId: "ns=2;s=RCV_CC394_SPS_.RCV_CC394_SPS.CC394_DEV_ENET",
    datatype: "UInt16",
    value: 0
};

return msg;`

I just wonder if you changed that in the update.

apbrauer73, 2024-10-16:
I wonder if the issue is with this code:

`if (msg.topic !== "writemultiple" && msg.topic !== "clearitems") {
  let opcuaDataValue = opcuaBasics.build_new_dataValue(msg.datatype, msg.payload);
  
  let item = {
    nodeId: msg.topic,
    datatype: msg.datatype,
    attributeId: opcua.AttributeIds.Value,
    indexRange: null,
    value: new opcua.DataValue({ value: opcuaDataValue })
  };
  
  item.value.sourceTimestamp = new Date(msg.timestamp).getTime();
  verbose_log("ITEM: " + stringify(item));
  
  writeMultipleItems.push(item); // This is where the item is added to the array.
}
`

This seems to build the array but I don't see how it will work  if the message topic is "writemultple".

I wonder if it should be like this....

`if (msg.topic === "writemultiple" && Array.isArray(msg.payload)) {
  msg.payload.forEach(payloadItem => {
    let opcuaDataValue = opcuaBasics.build_new_dataValue(payloadItem.datatype, payloadItem.value);
    
    let item = {
      nodeId: payloadItem.nodeId,
      datatype: payloadItem.datatype,
      attributeId: opcua.AttributeIds.Value,
      indexRange: null,
      value: new opcua.DataValue({ value: opcuaDataValue })
    };
    
    item.value.sourceTimestamp = new Date(msg.timestamp).getTime();
    verbose_log("ITEM: " + stringify(item));
    
    writeMultipleItems.push(item); // Add each item from the payload to writeMultipleItems
  });
}
`

MMBud21, 2024-10-16:
I also tested it. The one error
`"Client node error on: Write_Multiple error: ["Write multiple items session is not active!"]"`
that I described above is gone. It now works perfectly for me if I write anything in the topic, just not "writemultiple".

I think @apbrauer73 is right, it's just that the negation at writemultiple is wrong:
> if (msg.topic !== "writemultiple" && msg.topic !== "clearitems") {
>    let opcuaDataValue = opcuaBasics.build_new_dataValue(msg.datatype, msg.payload);

mikakaraila, 2024-10-16:
It will add msg into the nodesToWrite, normally it should add items one by one.
But if payload is an Array it expects that can be used. There is also topic clearitems to clean array.

MMBud21, 2024-10-16:
That sounds like a good solution.
In any case, the comparison `msg.topic !== "writemultiple"` must be adjusted to "===". Because currently sending works fine with every topic and without error messages, just not with the topic "writemultiple" :).

nikolaioak, 2024-10-16:
It seems along with this issue and subsequent fix (v0.2.334) there were a number of console.log() entries left for debugging within 102-opcuaclient.js and makes my Node-RED logs unusable since I use writemultiple for some heartbeat tags between systems. Can the logging be evaluated/removed?

apbrauer73, 2024-10-16:
I just tried changing the topic to something other than "writemultiple" and it worked like it did for @MMBud21.  @mikakaraila when you fix this also correct the status message, I noticed that when I trigger the multiple write the node is stuck with the status  "active writing".  

[image]

mikakaraila, 2024-10-16:
Cleanup console and set status to values written (green).
Comparison cannot be changed as it will break the other way to use writemultiple.

closed 2024-10-16 by mikakaraila

MMBud21, 2024-10-17:
Good morning @mikakaraila,
I just tested it with version 0.2.335.
[image]

It only works with all topics, just not with "writemultiple". But if you know that, it's not a problem. As you can see in the screenshots, it works perfectly with the topic "whateverIwant".

I think a lot of people are now using your client for some important applications and if basic functions like WriteMultiple no longer work after an update, that's obviously annoying. But of course we all know that you do this in your free time and certainly don't always have the time or inclination to fix the errors.
Thank you very much for your efforts and have a good day at work :).

Topic = "writemultiple"
[image]

Topic = "whateverIwant"
[image]

MMBud21, 2024-10-17:
Hey @mikakaraila,
I looked at your code for WriteMultiple in the 102-opcuaclient.js myself and am relatively sure that I have found the problem.

It currently works if we do not use the topic "writemultiple". Your comparisons above were correct, however, so at first I could not understand how this is possible. Then I changed the debug messages and found that we are not using the "new code" for communication.

Directly below there is another query that you called the "OLD original way". I also adjusted the debug messages there and lo and behold, this communication is currently being used. Because there is also no query there as to whether the topic is "writemultiple".

[image]

[image]
The data is written, I think the dot is just not green because you set the color somewhere when the text is "values ​​written". When I changed it to "values ​​written old" the dot is red, but the function is OK. I didn't look any further at where it sets the color, but it doesn't matter.

Then I looked into why the error "no items to write" always occurs with the topic "writemultiple". The problem is probably that you only fill the array in the upper part (red box, orange arrow), which is only processed if the topic is not "writemultiple" and payload is not an array.

In the lower part (green box) you first check whether the array has items. If that is not the case (length === 0), the message we get appears.

To test it, I deactivated the old part and filled the array in the green part with data before checking what length it has. After that it works perfectly with "writemultiple" but no longer works with all other topics, as it should.

[image]