mikakaraila/node-red-contrib-opcua #741: Write ExtensionObjet crash node red . previously worked on version 0.2.332

martinjulienp, 2024-10-21:
Write ExtentionObject crash node red: 

log : 21 Oct 16:12:25 - [red] Uncaught Exception:
21 Oct 16:12:25 - [error] TypeError: extensionObject.isArray is not a function
    at build_new_extensionObject_dataValue (/usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1660:50)
    at runMicrotasks (<anonymous>)
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
    at async write_action_input (/usr/src/node-red/node_modules/node-red-contrib-opcua/opcua/102-opcuaclient.js:1614:11)

i try this node : [
    {
        "id": "65cc1ba05eddc48f",
        "type": "inject",
        "z": "af0e3e1f0a75ac7a",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "",
        "once": false,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "x": 1250,
        "y": 140,
        "wires": [
            [
                "527f10dd84a24e25"
            ]
        ]
    },
    {
        "id": "527f10dd84a24e25",
        "type": "function",
        "z": "af0e3e1f0a75ac7a",
        "name": "write endCycle",
        "func": "msg.dataType = \"ExtensionObject\";\n// Merge new values to payload\nmsg.payload={\n    startCycle:false,\n    endCycle: true,\n    missionId: 123\n};\n// NOTE:\n// typeId need to constructExtensionObject\nmsg.topic = \"ns=2;s=CycleExc;datatype=ExtensionObject;typeId=ns=2;i=3004\";\nreturn msg;",
        "outputs": 1,
        "timeout": 0,
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 1500,
        "y": 140,
        "wires": [
            [
                "46e002d9b9d11a23",
                "2712cd1345a5994e"
            ]
        ]
    },
    {
        "id": "46e002d9b9d11a23",
        "type": "OpcUa-Client",
        "z": "af0e3e1f0a75ac7a",
        "endpoint": "b48779ba9fb6f2e5",
        "action": "write",
        "deadbandtype": "a",
        "deadbandvalue": 1,
        "time": 10,
        "timeUnit": "s",
        "certificate": "n",
        "localfile": "",
        "localkeyfile": "",
        "securitymode": "None",
        "securitypolicy": "None",
        "useTransport": false,
        "maxChunkCount": 1,
        "maxMessageSize": 8192,
        "receiveBufferSize": 8192,
        "sendBufferSize": 8192,
        "name": "Opc Ua Client Write",
        "x": 1730,
        "y": 140,
        "wires": [
            [],
            []
        ]
    },
    {
        "id": "b48779ba9fb6f2e5",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://opcuaserver:62540/OpcUaServeur",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
] ,  thank you in advance for your help. best regard

mikakaraila, 2024-10-22:
Hmm, I tried to extend functionality but I made a new bug, sorry.

mikakaraila, 2024-10-22:
Can you test this quick fix?

Line 1660:           if (extensionObject && extensionObject.length !== undefined) {

martinjulienp, 2024-10-24:
Hello,
 I'm sorry, I don't know how to test your fix.
 I'm using node red in a container and I'm getting your modul via npm.
 Someone else can test it. For the moment I have fixed the version to 0.2.332. I would be delighted to update when the new version is released. We also use the write ExtentionObject[], today I do not do the operation from node red, but I would be interested in the function.  Thank you for your help and your analysis. Best regard

mikakaraila, 2024-10-24:
Ok, I will put this to next version. I expect you can then build your container and test this...

mikakaraila, 2024-10-28:
Test with the latest version 0.2.337

closed 2024-10-28 by mikakaraila