mikakaraila/node-red-contrib-opcua #742: The local certificate file option is invalid

dingling1990, 2024-10-22:
I am using the local certificate file option, which will automatically generate its own certificate instead of using the local certificate，Is there something I set wrong? Looking forward to your reply

[image]
[image]

[image]

mikakaraila, 2024-10-22:
Try to use / instead of \

dingling1990, 2024-10-22:
> 尝试使用 / 代替 \

I changed it/ but it still doesn't work. I still use the default certificate. I can get the data after adding trust to the server.

If the file is wrong, there will be a prompt that pem is not found. Is there another problem?

mikakaraila, 2024-10-22:
Can you put here some log/console output? 
Otherwise I am blind and trying to guess problem.

Eldarion85, 2024-10-22:
I thought there was no longer an option to specify the path to the certificate?
Even in the field it says that it is no longer used.
I also use my own PKI and if the certificates are in the right folder, it works perfectly.

[image]

dingling1990, 2024-10-23:
> Can you put here some log/console output? Otherwise I am blind and trying to guess problem.

Use Local certificate ，The generated one is NodeOPCUAClient， Log/console output as shown

[image]

dingling1990, 2024-10-23:
> 我以为不再有指定证书路径的选项了？ 甚至在字段中也显示不再使用。 我也使用自己的 PKI，如果证书位于正确的文件夹中，它就可以完美运行。
> 

HI  Use Local certificate ，The generated one is NodeOPCUAClient，Can you tell me your PKI configuration path?
[image]

dingling1990, 2024-10-23:
> I thought there was no longer an option to specify the path to the certificate? Even in the field it says that it is no longer used. I also use my own PKI and if the certificates are in the right folder, it works perfectly.

HI Use Local certificate ，The generated one is NodeOPCUAClient，Can you tell me your PKI configuration path? tks
[image]

Eldarion85, 2024-10-23:
Hi,
I used to use my own path for the certificates until it just stopped working.
You can read about it in the issues:
https://github.com/mikakaraila/node-red-contrib-opcua/issues/577
https://github.com/node-opcua/node-opcua-crypto/issues/21
https://github.com/mikakaraila/node-red-contrib-opcua/issues/664

In any case, the result was that you can no longer specify your own path, but a fixed directory is used.
I don't know if this will help you, but in my Debian Linux the certificates have to be in the folder:
/home/YOURUSER/.config/node-opcua-default-nodejs/PKI

[image]

mikakaraila, 2024-10-24:
I expect his has changed as certificate manager was taken into use.
I will fix dialog text (no code changes).

mikakaraila, 2024-10-24:
Fix to dialog:
Path is not used anymore! Example: selfSigned.pem
Path is not used anymore! Example: private_key.pem

Info text:
Client will try to load server certificate from endpoint if server can provide it, copy server certificate under pki folder: 
USER-HOME-FOLDER/.config/node-opcua-default-nodejs/PKI/
Local file must be above folder and certificate filename can be then used to use server certificate (.pem ending file).

mikakaraila, 2024-10-28:
Check with latest version 0.2.337

closed 2024-10-28 by mikakaraila