mikakaraila/node-red-contrib-opcua #743: Write Multiple no longer works

Eldarion85, 2024-10-22:
There are two closed issues (#723 and #740), both of which are on the same topic "WriteMultiple". Both are closed (fixed), but they don't work.

I've now tested it again and looked at the code in the "102-opcuaclient.js". It is definitely the case that the processing for WriteMultiple runs via the part of the program that you called "OLD original way to use payload". Above that is the "new part", which is not used. However, there is a comment from you there saying "Do not try to run old way". In the current version, the old way is used permanently and no data can currently be written using the topic "writemultiple". See the last entry in #740.

mikakaraila, 2024-10-22:
As I tested with OPCUA-TEST-NODES.json (examples) it worked v0.2.335. 
Could you provide example that I can use for testing?

Eldarion85, 2024-10-22:
I looked at the examples for "Write Multiple" in your OPCUA-TEST-NODES.json. Your topic for the "ALL" function is "ns=5;s=Double" (see screenshot). But you always say it should be "writemultiple". If you set it to "writemultiple" you get the same error message as all of us.

[image]

I'm trying to understand how you even envision the function. Do you have to specify data first and then send a message without data, just with the topic "writemultiple"? I think we all have these problems because we don't want to collect data beforehand. We create a single message with an array (payload) of data (nodeId, datatype and value) and the topic is "writemultiple". Then the client should write this record immediately, just like it always worked.

I can also write data with Write Multiple in v0.2.335, but only if I set the topic to something other than "writemultiple". As soon as it is "writemultiple", the client reports that there is no items to write. This is correct, because the data to be written is contained in the same message as the topic "writemultiple" (see below).

[image]

mikakaraila, 2024-10-23:
Thanks for pointing this out, I will fix test case and code.
Business trip today... most probably I will get it done tomorrow.

mikakaraila, 2024-10-23:
OK was easy to fix

Eldarion85, 2024-10-23:
Good morning,
I tested the new version and Write Multiple now works again with the topic "writemultiple".

For testing purposes, I temporarily changed the logging in the "102-opcuaclient.js" in two places to find out which part of the script is being used. I only added "new" and "old" to the log.
[image]

If the topic is "writemultiple", the "new part" is used in the code:
[image]

If the topic is something other than "writemultiple", the "old part" is used further down.
[image]

I don't know if that's what you wanted, but since I've now set the topic to "writemultiple" everywhere, everything works great for me now. But no matter which topic you enter, it always works now. It's just processed in two different places ("new way" and "old way") in the code.

Thank you for the fix :).

closed 2024-10-28 by mikakaraila