mikakaraila/node-red-contrib-opcua #744: BadNodeIdUnknown (0x80340000)

Kromschu1234, 2024-10-22:
Hello everyone,

I just wanted to give a quick description of an issue we're encountering. We are receiving the error (BadNodeIdUnknown 0x80340000).

What we are trying to achieve is communication with a PLC via UaExpert. So far, the OPC server connection is working perfectly.

The problem started when we tried to connect the OPC server with a MySQL database using Node-RED. Everything works fine up till 
Function 4, but when the data is sent to the OPC client, we get the error code mentioned above.

In UaExpert (and also with the PLC), we are using an array table. I’m a newbie with this kind of setup. We've double-checked the NodeIDs in UaExpert multiple times, but there might be something we're overlooking.

What function 4 contains:

let result = msg.payload;
let payloadArray = [];

result.forEach((row, index) => {
    if (row.position && row.fuseType) {
        payloadArray.push({
            nodeId: `ns=3;s="Position_LAD/FBD"."point[${index}]"."position"`,
            datatype: "Int",  
            value: row.position
        });

        payloadArray.push({
            nodeId: `ns=3;s="Position_LAD/FBD"."point[${index}]"."fuseType"`,
            datatype: "Int",  
            value: row.fuseType
        });
    } else {
        node.warn(`Missing data in row at index ${index}. Row data: ${JSON.stringify(row)}`);
    }
});

if (payloadArray.length === 0) {
    node.error("No valid data to send to OPC UA Client");
    return null;
}

node.warn(`Generated payload array: ${JSON.stringify(payloadArray)}`);

msg.payload = payloadArray;
return msg;

Thanks for your advice!
[image]
[image]

Kromschu1234, 2024-10-24:
Right now everything is a good advice, thanks for it. Tomorrow in work I will try it .

mikakaraila, 2025-06-27:
I expect this can be closed

closed 2025-06-27 by mikakaraila