mikakaraila/node-red-contrib-opcua #745: Read multiple and return the valuein one message

YTCodeSpace, 2024-10-23:
I want to read multiple variables and return them in the same message, like an array. But after all i have tried, i always get them in separate messages. I used READ MULTIPLE but nothing changes.

[image]

mikakaraila, 2024-10-24:
I have to think a bit. I don´t want to break existing functionality, but this should be possible to implement.

YTCodeSpace, 2024-10-24:
> I have to think a bit. I don´t want to break existing functionality, but this should be possible to implement.

Now i need to use a Join function to combine the messages together. So it looks like that the options READ and READ MULTIPLE make no difference, and READ is even more easier since it doesn't need an extra inject to start. : )

mikakaraila, 2024-10-24:
I go it done, but I will test more.
There is now output pin 3 that will contain:

topic multipleItems[] and payload dataValues[]

Like this:
[image]

mikakaraila, 2024-10-28:
Output PIN 3 contains now array with dataValues, v0.2.337

closed 2024-10-28 by mikakaraila

JMBrgs, 2025-03-17:
The reading multiple in one object seems not to return values for string data types.

I checked an older version (0.2.298) , where writing msg.payload = all would return a single message with all data in a read multiple operation, and it didn't work with strings either.