mikakaraila/node-red-contrib-opcua #746: How to ignore the time discrepancy when building session

yinjilong, 2024-10-25:
I got the following error
```shell
02:24:12.092Z :client_secure_channel_layer   :804   [NODE-OPCUA-W33]  client : server token creation date exposes a time discrepancy late by 17 hours 4 minutes 28.660 seconds
 ...                                                remote server clock doesn't match this computer date !
 ...                                                 please check both server and client clocks are properly set !
 ...                                                 server time :2024-10-24T09:19:43.431Z
 ...                                                 client time :2024-10-25T02:24:12.091Z
 ...                                                 transaction duration = early by 0.000 seconds
 ...                                                 server URL = opc.tcp://192.168.1.201:4840
 ...                                                 token.createdAt  has been updated to reflect client time
my-PLC UA Client(SUB): Create session with userIdentity node.client false userIdentity {"type":0}
25 Oct 11:24:12 - [warn] [OpcUa-Client:my-PLC UA Client(SUB)] Create session with userIdentity node.client false userIdentity {"type":
```

There is no problem using other opcua client such as integration objects opcua client.

How can I let the nodered-opcua client also accept the connection even if clock doesn't match ?
Thank you.

mikakaraila, 2024-10-25:
Sorry, but there is no option available in the node-opcua to ignore time difference.
Time between server & client cannot differ more than 60 seconds. It is hard coded into the security checks.

yinjilong, 2024-10-25:
OK, I see, 
Thanks for your reply.

closed 2024-11-17 by yinjilong