mikakaraila/node-red-contrib-opcua #749: OPC-UA Subscription Stops Receiving Value Changes After ~4 Hours

softwareentwicklungsema, 2024-10-31:
I’m using Node-RED to read 10 variables from an OPC-UA server with the Subscribe method. These variables are subscribed with a 2-second interval and are configured to output on value change. This setup works as expected initially, but after approximately 4 hours, the subscribed variables stop updating upon value change.

Additional Information:

    Alarm events are also being monitored in parallel using the AlarmEvents method, and these continue to function correctly even after the 4-hour mark.
    Reading variables using Read or Read Multiple methods still works normally without any issues.

Environment:

    Node-RED version: 4.0.5
    OPC-UA server: Siemens Sinumerik One
    node-red-contrib-opcua: 0.2.337

Steps to Reproduce:

    Set up OPC-UA Subscribe method in Node-RED for 10 variables with a 2-second interval.
    Wait for approximately 4 hours.

Expected Behavior:
Subscribed variables should continue to update on value change indefinitely.

Actual Behavior:
After ~4 hours, subscribed variables stop updating on value changes.

Questions / Possible Causes:

    Is there a known issue with long-running subscriptions in Node-RED OPC-UA?
    Are there any recommended configurations to maintain subscription updates over extended periods?

Thank you for any assistance or insights!

nikso-serg, 2024-11-08:
A similar problem. Moreover, a different subscription method gives different results. 
If you do it through multiple, then the subscription is discounted after about 12 hours. I associate this with problems in the OPC server itself. So far, I have solved the problem by forcibly reconnecting as soon as time stops coming from the OPC server.
[image]

nikso-serg, 2024-11-08:
A similar problem. I attribute this to problems in the OPC server itself. So far, I have solved the problem by forcibly restarting the OPC client.

[image]

TheeeEnding, 2025-01-14:
Hi, I have similar problems with some OPCUA servers, would you mind sharing your flow how to restart the subscription @nikso-serg? :)
Thanks in advance!