mikakaraila/node-red-contrib-opcua #750: Is there everyone give me an example about "setUsers" command? thanks

xrq1051, 2024-11-01:


mikakaraila, 2024-11-01:
Hmm, belongs perhaps to node-opcua....

xrq1051, 2024-11-02:
Yes, it's a server node that belongs to node-red-contrib-opcua, but I didn't find the commit category when I submitted, so I just submitted it. I've tried so many times that I still can't add users according to the brief description, so I wondered if anyone could provide a complete example

mikakaraila, 2024-11-11:
Look users.json file and your msg.payload should contain similar JSON object with userName, password and role.

xrq1051, 2024-11-11:
Thank you, I solved that problem by analyzing the users.json later, but I encountered a problem later. After I saved the configuration through saveAddressSpace and loaded the saved file with loadAddressSpace, the configuration did not take effect.

获取 Outlook for iOS<https://aka.ms/o0ukef>

mikakaraila, 2024-11-11:
Any errors on the console?

xrq1051, 2024-11-11:
No error is reported, the console prompts the opcua server to restart, wait for the restart to complete, connect with the client, and find that the saved configuration is not effective

获取 Outlook for iOS<https://aka.ms/o0ukef>