mikakaraila/node-red-contrib-opcua #752: Bug Report: addFolder function fails on version 0.2.337

Meloman-zz, 2024-11-07:
Hello,

I've encountered an issue with the `addFolder` function in `node-red-contrib-opcua`. 

**The function works correctly in version 0.2.331 but fails in 0.2.337.** 

The error message received is simply "Error: Error", without additional details.

### Steps Taken

- I tested addFolder with my own code as well as using the provided sample flow from the documentation.
- In both cases, the same error appeared in version 0.2.337, while 0.2.331 functioned without issue.
- I have also tested version 0.2.337 with Node.js versions 18, 20, and 22, and with Node-RED versions 3.x and 4.x. The error persisted across all these configurations.

### Environment

- Node.js: 18, 20, and 22
- Node-RED: 3.x and 4.0.5

### Run Sequence with Screenshots

1. Sample executed on 0.2.331 :
[image]

2. Upgrade from palette :
[image]

3. Sample executed on 0.2.337 :
[image]

Additional Information

Before posting this issue, I tried to troubleshoot on the Node-RED forums to confirm the problem wasn’t due to a mistake on my part. You can find the discussion here: https://discourse.nodered.org/t/opcua-unable-to-execute-addfolder-opcuacommand/92918/12

Could you please investigate this issue?

Thank you!

mikakaraila, 2024-11-07:
Ack, I will take a look when I will have some extra time... perhaps tomorrow.

mikakaraila, 2024-11-11:
That part is now working, but another case needs still investigation, under work...

mikakaraila, 2024-12-09:
New version is available, re-open if not working.

closed 2024-12-09 by mikakaraila