mikakaraila/node-red-contrib-opcua #753: No automatic reconnect after longer OPC server downtime

vigh-barnabas, 2024-11-13:
I'm using 0.2.331 but tried it with 0.2.336 too.
After an OPC server is turned off for longer periods (like 5-6+ hours or days).
When the OPC server is turned back the OPC client node doesn't connect back.
It's either stuck in a reconnection loop (the nodes status shows reconnect xxx). I seen it in the hundreds.
Or the node status say BadConnectionClosed (this get spammed in the node-red debug window too)

I tired sending a disconnect and reconnect based on the status of the node but it dosn't work.
So far the only thing that work if I use the node-red action "Restart flows".
Any suggestion what should i try?

ruifilipefsilva, 2024-11-24:
Having similiar issues...
Please advise on how to handle better failover over BadConnectionClosed Errors and lack of a efective reconnection.

Also, can I ask, that similiar error change the node status to red.
I have lots of issues with nodes being green/yellow but node-red syslogs getting tons of connection errors.

Thank you

mikakaraila, 2024-11-25:
I expect this is due old version of node-opcua, I will check new version for the next release...

mikakaraila, 2024-11-25:
Hmm, perhaps I should take this as parameter into the client node as it seems to be sometimes true and sometimes false depending on server.

mikakaraila, 2025-06-27:
I expect this can be closed