mikakaraila/node-red-contrib-opcua #762: OPC UA 64bit Data reading problem.

DavidwWoo, 2024-12-10:
I just testing for mutireading function /w KEPServer  (node-red is running in mac /w M1 chip)

[image]

[image]

Ramp4 is getting correct value (it's 32bit data)
but Ramp5~Ramp7) get data little bit strange
AM i missing something? or set incorrectly?

I want to check my flow and guide from u!
Is bug or my mistake?

this is my testing flow

`[
    {
        "id": "b8398a925970a5fb",
        "type": "tab",
        "label": "플로우 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "2ab2468e8944774f",
        "type": "split",
        "z": "b8398a925970a5fb",
        "name": "",
        "splt": "\\n",
        "spltType": "str",
        "arraySplt": "1",
        "arraySpltType": "len",
        "stream": false,
        "addname": "",
        "x": 490,
        "y": 240,
        "wires": [
            [
                "7d9c88d37756f736"
            ]
        ]
    },
    {
        "id": "4f5f8233500f7c47",
        "type": "function",
        "z": "b8398a925970a5fb",
        "name": "OPC UA Read Array (24개)",
        "func": "//전체 노드 Id 목록\n\nmsg.payload=[\n\n      //Kepserver\n      // \"ns=2;s=Simulation Examples.Functions.Ramp1;datatype=Int32\",\n      // \"ns=2;s=Simulation Examples.Functions.Ramp2;datatype=Float\",\n      // \"ns=2;s=Simulation Examples.Functions.Ramp3;datatype=UInt16\",\n      \"ns=2;s=Simulation Examples.Functions.Ramp4;datatype=Int32\",\n      \"ns=2;s=Simulation Examples.Functions.Ramp5\",\n      \"ns=2;s=Simulation Examples.Functions.Ramp6\",\n      \"ns=2;s=Simulation Examples.Functions.Ramp7\"\n      // \"ns=2;s=Simulation Examples.Functions.Ramp8\",\n      // \"ns=2;s=Simulation Examples.Functions.Random1\",\n      // \"ns=2;s=Simulation Examples.Functions.Random2\",\n      // \"ns=2;s=Simulation Examples.Functions.Random3\",\n      // \"ns=2;s=Simulation Examples.Functions.Random4\",\n      // \"ns=2;s=Simulation Examples.Functions.Random5\",\n      // \"ns=2;s=Simulation Examples.Functions.Random6\",\n      // \"ns=2;s=Simulation Examples.Functions.Random7\",\n      // \"ns=2;s=Simulation Examples.Functions.Random8\",\n      // \"ns=2;s=Simulation Examples.Functions.Sine1\",\n      // \"ns=2;s=Simulation Examples.Functions.Sine2\",\n      // \"ns=2;s=Simulation Examples.Functions.Sine3\",\n      // \"ns=2;s=Simulation Examples.Functions.Sine4\",\n      // \"ns=2;s=Simulation Examples.Functions.User1\",\n      // \"ns=2;s=Simulation Examples.Functions.User2\",\n      // \"ns=2;s=Simulation Examples.Functions.User3\",\n      // \"ns=2;s=Simulation Examples.Functions.User4\"\n\n      //prosys opcua server\n      // \"ns=3;i=1007\",\n      // \"ns=3;i=1001\"\n]\nreturn msg;",
        "outputs": 1,
        "timeout": "",
        "noerr": 0,
        "initialize": "",
        "finalize": "",
        "libs": [],
        "x": 280,
        "y": 240,
        "wires": [
            [
                "2ab2468e8944774f"
            ]
        ]
    },
    {
        "id": "9834cc89180fb8cd",
        "type": "inject",
        "z": "b8398a925970a5fb",
        "name": "Start",
        "props": [
            {
[... 185 lines cut ...]
        "name": "",
        "x": 620,
        "y": 380,
        "wires": [
            [
                "cdccab90eb9df7d1"
            ],
            [],
            []
        ]
    },
    {
        "id": "85c4164544a8a87a",
        "type": "OpcUa-Endpoint",
        "endpoint": "opc.tcp://192.168.11.14:49320",
        "secpol": "None",
        "secmode": "None",
        "none": true,
        "login": false,
        "usercert": false,
        "usercertificate": "",
        "userprivatekey": ""
    }
]
`

mikakaraila, 2024-12-10:
As JavaScript does not support 64bit values it is divided into two bytes as you see in payload.

mikakaraila, 2024-12-10:
You can try to convert two bytes into 32bit as long as it does not overflow solution would work.

DavidwWoo, 2024-12-10:
> As JavaScript does not support 64bit values it is divided into two bytes as you see in payload.

Is that so? I thought JavaScript also supports 64-bit numbers. There’s a type like BigInt, isn’t there?

mikakaraila, 2024-12-11:
You can combine two bytes and use these:
There are a few JS libraries around that can handle larger numbers, e.g., [BigNumber](http://jsfromhell.com/classes/bignumber) and [Int64](https://github.com/broofa/node-int64).

As there only a few cases that really needs 64bit numbers, sorry.

Steve-Mcl, 2025-03-03:
> As JavaScript does not support 64bit values it is divided into two bytes as you see in payload.

JavaScript does handle 64 bit numbers (natively, without any libs) via the primitive `BigInt`.  Ref: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/BigInt#description