mikakaraila/node-red-contrib-opcua #766: Why was the great improvement of the different subscriptions removed again?

Eldarion85, 2024-12-27:
I saw that in the new version you register the subscription individually on the server. Here is a screenshot from the Siemens PLC. Before, there was only one entry (yellow frame) with the total number of all variables (173). Now I can also see the individual subscriptions in the server (green frame), which is a very good improvement.
Thank you very much for that!
Now I just have to solve the problem with Writemultiple :).

[image]

_Originally posted by @Eldarion85 in https://github.com/mikakaraila/node-red-contrib-opcua/issues/723#issuecomment-2385716834_

Eldarion85, 2024-12-27:
In the current version, all subscriptions that are subscribed to one after the other via the same client node are summarized again.
Recently there was a version where this was different and behaved as it should. This helped enormously with the evaluation on the server and was a very good improvement.

mikakaraila, 2025-03-31:
Could you be more specific, I just don´t get your idea/issue...