mikakaraila/node-red-contrib-opcua #771: Error connecting to OPCUA

InakiIribarren, 2025-01-17:
Hi,

we are trying to connect nodered (installed in redhat 8) to an opcua server. 

>[info] Node-RED version: v1.3.5
>[info] Node.js  version: v18.20.5
>[info] Linux 4.18.0-305.7.1.el8_4.x86_64 x64 LE
 >node-red-contrib-opcua@0.2.339

We have tried a lot and finally our last attempt we have the next error:

```  
Node-RED[4681]: 07:03:52.786Z :client_secure_channel_layer   :1057  ERROR !!! callback has thrown en error  TypeError: Cannot read properties of undefined (reading 'createdAt')
  at hasTokenReallyExpired (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/token_stack.js:27:39)
  at TokenStack.getTokenDerivedKeys (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/token_stack.js:88:13)
  at Object.getDerivedKey (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/token_stack.js:41:32)
  at ClientSecureChannelLayer._ClientSecureChannelLayer__get_security_options_for_MSG (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1315:123)
  at ClientSecureChannelLayer._ClientSecureChannelLayer__get_security_options (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1336:161)
  at ClientSecureChannelLayer._ClientSecureChannelLayer__sendSecureOpcUARequest (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1350:169)
  at ClientSecureChannelLayer._ClientSecureChannelLayer__internal_perform_transaction (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1133:127)
  at ClientSecureChannelLayer._ClientSecureChannelLayer__performMessageTransaction (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:1088:133)
  at ClientSecureChannelLayer.performMessageTransaction (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:371:134)
  at OPCUAClientImpl.performMessageTransaction (/home/admitotgat/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:771:29)
  at OPCUAClientImpl.getEndpoints (/home/admitotgat/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:819:14)
  at OPCUAClientImpl.getEndpoints (/home/admitotgat/.node-red/node_modules/thenify/index.js:65:46)
  at OPCUAClientImpl.getEndpoints (/home/admitotgat/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:804:25)
  at OPCUAClientImpl.getEndpoints (/home/admitotgat/.node-red/node_modules/thenify/index.js:65:46)
  at /home/admitotgat/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:529:26
  at /home/admitotgat/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:3675:28
  at replenish (/home/admitotgat/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:447:21)
  at iterateeCallback (/home/admitotgat/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:431:21)
  at /home/admitotgat/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:328:20
  at /home/admitotgat/.node-red/node_modules/node-opcua-client/node_modules/async/dist/async.js:3680:17
  at /home/admitotgat/.node-red/node_modules/node-opcua-client/dist/private/client_base_impl.js:521:21
  at ClientSecureChannelLayer.<anonymous> (/home/admitotgat/.node-red/node_modules/node-opcua-secure-channel/dist/source/client/client_secure_channel_layer.js:833:9)
```  

After this error keep trying to connect and finally give this error [error] [OpcUa-Client:234585fc.0451fa] Invalid endpoint. 

We have put the opcua server certificates in the folder /root/.config/node-opcua-default-nodejs/PKI/issuers/
And the client certifcates in  /root/.config/node-opcua-default-nodejs/PKI/own/

Somebody can help with this issue. Best regards!

IbfUwe, 2025-01-17:
Hallo ,

I Think your Node-Red installation is too old.
Actual I have installed 4.0.5
Actual Version is 4.0.8

mikakaraila, 2025-03-31:
Check that server and client have same current time!! Are you uses NTP?