mikakaraila/node-red-contrib-opcua #772: Subscriptions

rmv5004, 2025-01-17:
It does not seem like there is any prevention from subscribing to an item multiple times.  Is there a way to check active subscriptions so I don't subscribe more than 1 time?  Should there be prevention from subscribing to the same item?  Eventually it seemed to overload and crash the server.

VitorCaldana, 2025-02-05:
I noticed there is a restriction for submissions on the server. I can only get 10 variables per client and a total of 100 variables.

I believe this is set up on the code, however it is not possible to modify it. Shouldn´t this be a configurable parameter on the server?
If so, i kndly ask that an update is provided. If not, could someone explain to me why that is the case.

Thanks!

mikakaraila, 2025-03-31:
The latest OPC UA specification v1.05 contains server limits. I am not sure if your server uses these.
Node-red embedded OPC UA Server can setup limits, but I am not sure if clients can use limits... sorry.

rmv5004, 2025-10-02:
This seemed to be linked to a Codesys OPC-UA server issue.  Codesys recently fixed several OPC-UA server bugs.