mikakaraila/node-red-contrib-opcua #777: Subscription Limitation

VitorCaldana, 2025-02-06:
Dear Developer.

I´m a PhD student and am using Node-RED and your library to connect to a Flexible Manufacturing System (FMS) with 300+ variables. Each working station of said FMS has between 40 and 60 variables that need to be monitred via subscription (since I´m using Asset Adminsatration Shell) and I found a limitation to the Server Block on the library.

The server won´t allow me to get more than 100 variables subscriped, and 10 per session or OPC UA Client.

Looking at the coding files, I found the following at server_capabilities.js

    maxSubscriptions: 100,
    maxMonitoredItems: 1000000, // 1 million
    maxSubscriptionsPerSession: 10,
    maxMonitoredItemsPerSubscription: 100000, // one hundred thousand
    maxSelectClauseParameters: 100,
    maxWhereClauseParameters: 100,
    maxMonitoredItemsQueueSize: 60000,

I tried changing those numbers to values that would make my application run, however even after editing anf saving the file, restarting Node-RED, restarting the computer and removing and reinserting the Server node I get all the confirmation message from subscription from the client butthe number o SubscriptionsPerSession is still 10. I tried to find the variable at other files, but VSCode can´t find them anywhere else.

I noticed that these settings are not visible at the UI, is this why it is not working??

[image]

mikakaraila, 2025-03-31:
These settings are for the Node-Red OPC UA Server not for the FMS.
FMS Server limits subscriptions/items to optimize resource usage.
You can create multiple subscriptions and add only 50 items/subscription.

VitorCaldana, 2025-04-10:
Thanks for the update.

closed 2025-04-10 by VitorCaldana