mikakaraila/node-red-contrib-opcua #779: Writemultiple is not working / has no reaction

YTCodeSpace, 2025-02-12:
I'm using an Inject with msg.topic="writemultiple", after that there is a function with nodeID and values like:

`msg.payload = [{
   nodeId: "ns=2;s=unit/ID1",
   datatype: "Double",
   value: 562.5
},
   {
      nodeId: "ns=2;s=unit/ID2",
      datatype: "Int16",
      value: 12
   }]
return msg;
`
When i click inject, nothing happens, no error, no warning, only a feedback from browser says inject done. 

Write one item works fine btw.

mikakaraila, 2025-03-31:
Try to first inject nodeIds and then separately topic/action...

TecniciCsb, 2025-04-07:
Hi, 

have the same problem. "readmultiple" works fine but writemultiple gives some problems.

If i use the following code

msg.topic = [{ nodeId: "ns=4;i=6262", 
                    datatype: "String", 
                    value: "0102145721245555" }, 
                { nodeId: "ns=4;i=6264", 
                    datatype: "String", 
                    value: "MULTIPLE WRITE TEST" }]; 

msg.topic = "writemultiple";

i get a "valure written" message from the OpcUa client node but i see no values written on PLC.

If i inject first the topic with writemultiple it says that there's no values to be written.
If i inject first the payload it returns that the node can't work with msg.topic empty

thanks for your help

TecniciCsb, 2025-09-30:
Hi, 

is the problem solved? I've not had the occasion to try it again...

Thank you

mdameda, 2025-12-03:
It's not already solved. I saw that into the function :

`async function writemultiple_action_input`  the msg.payload  coming from nodered component is not correct