mikakaraila/node-red-contrib-opcua #780: Node-red-contrib-opcua crashes when restarting flows

jeddors, 2025-02-12:
Hi,

If I restart my flows, node-red crashes with the following trace:
`12 Feb 21:28:33 - [info] Stopping flows
12 Feb 21:28:34 - [info] Stopped flows
12 Feb 21:28:34 - [red] Uncaught Exception:
12 Feb 21:28:34 - [error] Error: Already disposing
    at OPCUACertificateManager.<anonymous> (/root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:464:23)
    at Generator.next (<anonymous>)
    at /root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:31:71
    at new Promise (<anonymous>)
    at __awaiter (/root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:27:12)
    at OPCUACertificateManager.dispose (/root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-pki/dist/pki/certificate_manager.js:462:16)
    at OPCUACertificateManager.<anonymous> (/root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:58:38)
    at Generator.next (<anonymous>)
    at /root/.node-red/node_modules/node-red-contrib-opcua/node_modules/node-opcua-certificate-manager/dist/certificate_manager.js:8:71
    at new Promise (<anonymous>)
root@siemensiot:~#`

My software versions:
`Node-RED v4.0.8
Node.js v22.14.0
Linux 5.10.158-cip22-rt9+ind1 arm64 LE`
`npm -v    11.1.0`

I tried updating the node-red-contrib-opcua to version 0.2.339 but it did not fix the problem.

jeddors, 2025-02-12:
I think I have fixed it by downgrading to Node.js version 16.20.2, will see if it continues to crash.

0rion447, 2025-02-25:
I was having that certificate error as well. I solved it by downgrading as described in my issues error `#773`. I believe the certificate issue is rooted in a new CVE with the encryption algorithm that the `node-opcua` package uses. I think you can tell node red to ignore CVE issues at start up but downgrading was my choice as well.