mikakaraila/node-red-contrib-opcua #787: TypeError: crypto_utils.readCertificate is not a function - Heidenhain Certificates

florianbold, 2025-03-07:
Hello,

im currently trying to connect to a Heidenhain OPC UA Server instance on my machine. 

I have generated a client certificate on the HeidenHain Machine. The Alternative names for the machine are these:

URL=urn:E314949-OPSV5.domain.local:NodeOPCUA-Client
DNS-Name=E314949-OPSV5.domain.local

I have generated a user-certificate, which I know works because I could access the server via UAExpert with the same User certificate and put the Servers Exported certificate in my PKI/trusted certificates.

Now when i try to connect i get said "TypeError: crypto_utils.readCertificate is not a function" Error in my console.

Is this a known problem or am I doing something wrong? I have tried to look at closed issues regarding this problem but couldn't find a solution.

Here is the structure of my %appdata%\nodered-opcua-nodejs folder:

└───Config
    ├───PKI
    │   └───own
    │       ├───certs
    │       │       opcua-client-application-cert-20250307-130814.der
    │       │       opcua-client-application-cert-20250307-130814.pem
    │       │
    │       └───private
    │               opcua-client-application-cert-20250307-130814.key
    │               opcua-client-application-cert-20250307-130814.key.pem
    │
    ├───ServerPKI
    │   │   mutex.lock
    │   │
    │   ├───issuers
    │   │   ├───certs
    │   │   └───crl
    │   ├───own
    │   │   │   openssl.cnf
    │   │   │
    │   │   ├───certs
    │   │   │       certificate.pem
    │   │   │       certificate.pem.mutex
    │   │   │
    │   │   └───private
    │   │           private_key.pem
    │   │
    │   ├───rejected
    │   └───trusted
    │       ├───certs
    │       │       heopcua-rootca_heros5-69158_2025jan27_144102.pem
    │       │       heopcua-server_heros5-69158_2025jan27_144102.pem
    │       │       Node-RED OPCUA@E314949-OPSV5[c47af0f3b5a7797f7ed8afff4ae520fa0aa1d42f].pem
    │       │
    │       └───crl
    │               heopcua-rootca_heros5-69158_2025jan27_144104.crl
    │
    └───UserPKI
        │   mutex.lock
        │
        ├───issuers
        │   ├───certs
        │   └───crl
        ├───own
        │   │   openssl.cnf
        │   │
        │   ├───certs
        │   │       opcua-user-cert-20250305-104344.pem
        │   │
        │   └───private
        │           opcua-user-cert-20250305-104344.key.pem
        │
        ├───rejected
        └───trusted
            ├───certs
            └───crl

Thank you for any help regarding this issue

Mancio3096, 2025-03-28:
I'm the same problem

Mancio3096, 2025-04-15:
@mikakaraila

jeffsparks, 2025-05-08:
I don't exactly have a solution but to add information after struggling through it myself to help the next guy.

It seems the error is coming from the node-opcua, I guess the function was removed. In the commit for node-red-contrib-opcua 0.2.339, the node-opcua dependency changed from ^2.124.0 to ^2.138.1 so I'm guessing it was removed somewhere in between, but I have no idea.

I got around this error by using an older version. For example:

1. Navigate to the .node-red directory
2. Remove node-red-contrib-opcua and node-opcua
`node remove node-red-contrib-opcua`
`node remove node-opcua`
3. Manually install node-opcua 2.124 FIRST
`npm install node-opcua@2.124.0`
4. Manually install node-red-contrib-opcua 0.2.337
`npm install node-red-contrib-opcua@0.2.337`
5. Check to make sure it is the old version of node-opcuaIf you install node-opcua 2.124.0
`npm list node-opcua`

However... I found out for myself I don't use certificates for authentication anyways. In the older versions of this library there are better error messages which helped me realize this. On a clean node-red install using the latest version (0.2.339) with the latest node-opcua (2.153.0), using user credentials for the server, and specifying the certificates in the OpcUa-Client node, I do not get this error and have no issues.

Best of luck

username-simple, 2025-06-19:
Hey, probably a stupid question, but how do you install node-opcua-crypto?
I tried `npm install node-opcua/node-opcua-crypto` but still hae the same error.

closed 2025-06-27 by mikakaraila