mikakaraila/node-red-contrib-opcua #790: browseName is included in nodeId when adding OPC-UA variables

Meloman-zz, 2025-03-18:
**Description**
I'm using node-red-contrib-opcua to dynamically create an OPC-UA structure with folders and variables. The folders and hierarchy are working correctly, but I noticed that when I add a variable, the browseName gets included in the nodeId, which seems incorrect.

Example of the issue, when I add a variable like this:

```
messages.push({
    ...msg,
    payload: { opcuaCommand: "addVariable" },
    topic: `ns=1;s=FESP_100140.Electricity.Phase1.ActivePower;browseName=ActivePower;datatype=Float`
});
```
It gets created with:

    _NodeId: ns=1;s=FESP_100140.Electricity.Phase1.ActivePower;browseName=ActivePower
    BrowseName: 1:ActivePower
    DisplayName: ActivePower_

[image]

I would expect the NodeId to remain _ns=1;s=FESP_100140.Electricity.Phase1.ActivePower_ without the _browseName_ appended to it.

**Expected Behavior**
- The NodeId should only contain the unique identifier.
- The BrowseName should be a separate property, not appended to the NodeId.

**Environment**
    Node-RED version: 4.0.9
    node-red-contrib-opcua version: 0.2.339
    OPC-UA Client: Prosys OPC UA Browser
    Node.js version: 22.
    Operating System: [your OS]

**Questions**
- Is this the expected behavior, or is it a bug?
- If this is incorrect, is there a workaround to properly assign browseName without it being part of the NodeId?

Meloman-zz, 2025-03-19:
For now, here is a workaround to update the variables. 

I'm calling them like this:

```
{
  "payload": {
    "messageType": "Variable",
    "namespace": 1,
    "variableName": "FESP_100140.Default.Outlet.InternalComputers.BreakerFault;browseName=BreakerFault",
    "datatype": "Boolean",
    "variableValue": true
  },
  "_msgid": "6259413d945e3bc5",
  "_event": "node:d8f0fcbcb237cc87",
  "topic": "archivage_defaut.bdefdisj6prises"
}
```

Using:
_FESP_100140.Default.Outlet.InternalComputers.BreakerFault;browseName=BreakerFault_

Instead of:
_FESP_100140.Default.Outlet.InternalComputers.BreakerFault_

And it works.

closed 2025-03-31 by mikakaraila

Meloman-zz, 2025-04-01:
Why close this issue @mikakaraila ?
a workaround is not a fix !!!

mikakaraila, 2025-04-01:
I just don´t have time to fix these now.
You can try topic in different order of field, leave browseName as last one.
I expect parser part of the code is not optimal...