mikakaraila/node-red-contrib-opcua #791: High frequency re-connects of read-multiple that causes high load on a Siemens PLC

HansDamph, 2025-03-18:
Hello Mika,

first: Thank you for all the work an patience you have will all the users!

I have a application, that reads based on a trigger 340 variables of a PLC with READ MULTIPLE.
This works for most of the time very well and I receive the data after 1-3 seconds - which is ok.

The machine does not run all the time and when I go back, I see this strange behaviour:
https://github.com/user-attachments/assets/1077f28f-7bf8-47bd-8cfb-644babf64c7c

I can't get out of this situation but killing the node-red instance (restart the docker container). Disabling, Enabling the node does not affect the issue at all. All other nodes (read, write) do not show this. 

The problem is now, that this blocks other connections of that PLC because it holds more connections open. Any Idea what happens?

The Issue is on 3 different PLCS (Siemens 1516) running in individual docker containers.

Many thanks in advance!

HansDamph, 2025-03-21:
I could narrow it a litte bit down:
A longer time ago, the node lost the connection to the OPC-UA server. When I detect, that I do not receive a new message after a certain time, I use msg.action = reconnect to restart it.
For me it looks like, the reconnect creates a new connection, but does not terminate an existing one. Could that be the case?

mikakaraila, 2025-03-31:
Have to check this, but can you read server state? I mean if you server will tell to the client that it will go down in let´s say after 10 second you could shutdown client and then use cron to start it again after known shutdown period.

mikakaraila, 2025-03-31:
Sorry I pushed wrong button by mistake,