mikakaraila/node-red-contrib-opcua #793: Restarting all nodered flows via API cause "Client not yet created & connected" error

FrankRobot, 2025-03-27:
Hi, I've many connections problems between a Panasoni Panel, with onboard opcua Server, and a windows 10 MiniPC with nodered. It works well for hours, but when the network goes down, I've many reconnections attemps or Badtoomanysessions errors. To solve thiese issues  I've create a reset button, that reload all the flows via rest api. But now I receive many "reconnect" attempts and this new error: Client not yet created & connected!
Thanks in advance.
Frank

mikakaraila, 2025-03-31:
If you have multiple client nodes you can actually try to use one, but you have to add / inject msg.action into the message object.

FrankRobot, 2025-03-31:
Hi Mika, please give me an example. I'd have to read many data and then lauch different actions for different data. Or I've to write different data based on previous actions. Example: if a variable contain "1" I've to write "1" in another. Or if a data reach a predefined limit I've to write 1 to multiple data, etc... Thanks.